google/xls #1600: Token arrays needed to supplement channel arrays

mikex-oss, 2024-09-11:
### What's hard to do? (limit 100 words)

Trying to take advantage of channel arrays now that #704 is fixed.

However, I think I'm blocked by the fact that I can't keep track of tokens across the channel array separately. The test example as part of delivering the feature only shows sequential tokens:

https://github.com/google/xls/blob/b117a11ef67534bba97f8562b024278fb9ba7ee5/xls/dslx/ir_convert/proc_config_ir_converter_test.cc#L305-L311

However, if I'd like each channel in the array to operate in parallel, I want to save all those tokens so that I can sequence downstream sends/receives based on the specific token element.

I tried out token arrays and it first fails on:

```
F0911 15:12:26.916626    2400 statusor.cc:99] Attempting to fetch value instead of handling error INVALID_ARGUMENT: Type "token" has no defined signedness.
*** Check failure stack trace: ***
    @     0x564c153d0304  absl::log_internal::LogMessage::SendToLog()
    @     0x564c153cfe76  absl::log_internal::LogMessage::Flush()
    @     0x564c155a1b4e  absl::log_internal::LogMessage::~LogMessage()
    @     0x564c1539fa9f  Google3AbseilInternalLog::Hook()
    @     0x564c153d4618  absl::internal_statusor::ThrowBadStatusOrAccess()
    @     0x564c1352e896  xls::dslx::BuiltinTypeAnnotation::GetSignedness()
    @     0x564c1221ba73  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleArrayTypeAnnotation()
    @     0x564c122165a1  xls::dslx::Deduce()
    @     0x564c121eb0d4  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x564c122b4c7b  xls::dslx::DeduceCtx::Deduce()
    @     0x564c1221d9e5  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleTupleTypeAnnotation()
    @     0x564c122165a1  xls::dslx::Deduce()
    @     0x564c121eb0d4  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x564c122b4c7b  xls::dslx::DeduceCtx::Deduce()
    @     0x564c12216982  xls::dslx::DeduceAndResolve()
    @     0x564c12226e31  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleUnrollFor()
```

I tried hacking `BuiltinTypeAnnotation` to define a signedness for tokens, but then it just trips over thinking that it is a `sN` or `uN` type:

```
TypeInferenceError: uN[8] Want argument 0 to 'update' to be an array; got uN[8]
```

### Current best alternative workaround (limit 100 words)

I'm not aware of any workaround aside from keeping the sends/receives manually unrolled.

### Your view of the "best case XLS enhancement" (limit 100 words)

Would like to be able to do something like:

```
        let send_toks = unroll_for!(i, send_toks): (u32, token[N]) in u32:0..N {
            update(send_toks, i, send(tok, chan[i], x))
        }(zero!<token[N]>());
```

Could something like how we handle channel arrays be applied to tokens as well?

mikex-oss, 2024-10-25:
Related: https://github.com/google/xls/issues/691

The notable additional feature mentioned there is the ability to `join` on such a token array.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Token arrays (`token[N]`) are now supported in the DSLX typechecker and `ArrayType::HasToken()` (`xls/dslx/type_system/type.h:682`), with regression test `TokenArrayTypeAnnotation` in `xls/dslx/type_system/proc_typecheck_test.cc:30-33`. (Closed by Gemini)

closed 2026-11-14 by allight