google/xls #1615: Indefinite parsing and type-checking loops for parameterized procs

lpawelcz, 2024-09-19:
**Describe the bug**
While working on https://github.com/google/xls/pull/1613, we encountered an issue linked to https://github.com/google/xls/commit/afc720d6a35616c894b6aa342c49554afe0f35cd, which appears to cause indefinite parsing and type-checking loops for certain files, such as `mem_writer.x` (provided in this PR).

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout on the [branch](https://github.com/antmicro/xls/tree/axi) from https://github.com/google/xls/pull/1613
2. Execute the command `bazel build -c opt //xls/modules/zstd/memory:mem_writer_verilog`
3. Observe indefinite loop in parsing and type-checking `mem_writer` DSLX files:
```
    Converting DSLX file to XLS IR: xls/modules/zstd/memory/mem_writer.x; 48s remote-cache, processwrapper-sandbox
    Parsing and type checking DSLX source files of target mem_writer_dslx; 48s remote-cache, processwrapper-sandbox
```
4. Cancel bazel build
5. Revert https://github.com/google/xls/commit/afc720d6a35616c894b6aa342c49554afe0f35cd
6. Repeat the bazel build command: `bazel build -c opt //xls/modules/zstd/memory:mem_writer_verilog`
7. Observe correct IR and verilog generation for the `MemWriter` proc.

**Expected behavior**
Toolchain should be able to generate IR for the provided `MemWriter` proc .

**Additional context**
This is a blocker for https://github.com/google/xls/pull/1613 and as a consequence also for https://github.com/google/xls/issues/1211.

richmckeever, 2024-09-30:
What's happening here is that a couple of bugs, which are inconsistent sequencing of the parametric pattern "AxiStream<DATA_W, DEST_W, ID_W, DATA_W_DIV8>" in the zstd code, are exposing parametric env leakage in type checking, where we build maps like:

DEST_W => DATA_W_DIV8
DATA_W => DATA_W
DATA_W_DIV8 => ID_W
ID_W => DEST_W

and then try to figure out what e.g. DATA_W_DIV8 is, using such a map.

The faulty "AxiStream<wrong-order parametrics>" are in axi_writer.x and mem_writer.x. You may be able to get unblocked by fixing those to be consistent with the struct declaration. In the meantime I will work on a fix in the type checker.

lpawelcz, 2024-10-01:
@richmckeever thank you so much for pointing the errors in the sequencing of the parameters! I was able to fix those and generate the IR without problems.

commit 9081e204 (2024-10-02): Fix infinite loop in type checking with inverted parametric name order.

Context: https://github.com/google/xls/issues/1615

The main scenario we are dealing with here is:

```
struct Foo<X, Y, Z = {X}> { ... }

fn bar<X, Y>() -> Foo<Y, X> {
  // Reverse-ordered parametric names in the instantiation vs. the struct def.
  Foo<Y, X> {...}
}
```

The defaulted `Z` is not needed to hit the bug, but the bug is fallout from previously fixing that, and we want to keep it fixed.

In this scenario, it's OK to evaluate `Z` with the parametric env that has `X` and `Y` from the immediate context in `Concr [... cut]
closed 2024-10-02 by copybara-service[bot]