google/xls #1624: FUZZER: eval_proc_main crash 0330

allight, 2024-09-23:
**Describe the bug**
Fuzzer crash found on 9-23 hash 0330

crashers/crasher_2024-09-23_0330.x

Seems to timeout, possibly in LLVM.

allight, 2024-09-23:
This seems to be an IR that llvm just takes an inordinately long time to process...

Setting llvm opt level to 0 results in finishing in <1 second.

ericastor, 2024-09-30:
Profiling indicates that LLVM might be getting stuck in the DAGCombiner; it's spending 99.97% of the first two minutes inside `DAGCombiner::Run`. I might try to dump the LLVM IR & see if I can get the issue reported upstream, especially if I can reduce it at all.

allight, 2026-11-10:
This crasher (crasher_2024-09-23_0330.x) is no longer in the known-failing lists of //xls/fuzzer:regression_tests, so it appears to be fixed (re-enabled in https://github.com/google/xls/commit/793d3dea05a26072facee992d0151f1a0c81d943).

closed 2026-11-10 by allight