google/xls #1625: DSLX Language Server Build Error

selimsandal, 2024-09-24:
**Describe the bug**
The build fails when compiling the language server on macOS.

**To Reproduce**
- Compile DSLX language server on macOS

**Environment (this can be helpful for troubleshooting):**
 - OS: macOS Sequoia

**Error log**
```
external/com_google_absl/absl/container/internal/raw_hash_set.h:3088:17: error: type 'hasher' (aka 'absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>') does not provide a call operator
 3088 |     auto hash = hash_ref()(key);
      |                 ^~~~~~~~~~
external/com_google_absl/absl/container/internal/raw_hash_set.h:2836:20: note: in instantiation of function template specialization 'absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::find_or_prepare_insert<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>' requested here
 2836 |       auto res = s.find_or_prepare_insert(key);
      |                    ^
external/com_google_absl/absl/container/internal/container_memory.h:140:10: note: in instantiation of function template specialization 'absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::InsertSlot<true>::operator()<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, const std::piecewise_construct_t &, std::tuple<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>> &>, std::tuple<const std::unique_ptr<xls::dslx::BytecodeFunction> &>>' requested here
  140 |   return std::forward<F>(f)(key, std::piecewise_construct, std::move(p.first),
      |          ^
external/com_google_absl/absl/container/internal/container_memory.h:207:27: note: in instantiation of function template specialization 'absl::container_internal::memory_internal::DecomposePairImpl<absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::InsertSlot<true>, const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>> &, std::tuple<const std::unique_ptr<xls::dslx::BytecodeFunction> &>>' requested here
  207 |   return memory_internal::DecomposePairImpl(
      |                           ^
external/com_google_absl/absl/container/flat_hash_map.h:591:38: note: in instantiation of function template specialization 'absl::container_internal::DecomposePair<absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::InsertSlot<true>, const std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>> &>' requested here
  591 |     return absl::container_internal::DecomposePair(std::forward<F>(f),
      |                                      ^
external/com_google_absl/absl/container/internal/hash_policy_traits.h:134:15: note: in instantiation of function template specialization 'absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>::apply<absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::InsertSlot<true>, const std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>> &>' requested here
  134 |     return P::apply(std::forward<F>(f), std::forward<Ts>(ts)...);
      |               ^
external/com_google_absl/absl/container/internal/raw_hash_set.h:2439:26: note: in instantiation of function template specialization 'absl::container_internal::hash_policy_traits<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>::apply<absl::container_internal::raw_hash_set<absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>, absl::hash_internal::Hash<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::equal_to<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>>, std::allocator<std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>>::InsertSlot<true>, const std::pair<const std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>> &, absl::container_internal::FlatHashMapPolicy<std::tuple<const xls::dslx::Function *, const xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>>, std::unique_ptr<xls::dslx::BytecodeFunction>>>' requested here
 2439 |     return PolicyTraits::apply(InsertSlot<true>{*this, std::move(*slot)}, elem);
      |                          ^
[... cut ...] onst xls::dslx::TypeInfo *, std::optional<xls::dslx::ParametricEnv>> &, std::unique_ptr<xls::dslx::BytecodeFunction>, 0>' requested here
   46 |     cache_.emplace(key, std::move(bf));
      |            ^
5 errors generated.
Target //xls/dslx/lsp:dslx_ls failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 409.037s, Critical Path: 26.93s
INFO: 2661 processes: 1253 internal, 1408 darwin-sandbox.
ERROR: Build did NOT complete successfully

```

cdleary, 2024-09-24:
@selimsandal FYI there are some release binary builds for OS X (both aarch64 and x64) available on this branch (build via GH actions): https://github.com/xlsynth/xlsynth/releases/tag/v0.0.88 -- workflow is https://github.com/xlsynth/xlsynth/blob/cdleary/release-dylib-workflow/.github/workflows/build-and-release-dylib.yml

selimsandal, 2024-09-24:
@cdleary thank you for the link, I didn't see that repo.

I also wanted to ask about guidance on creating delay models for other PDKs / FPGAs. Should I open an issue or discussion on GitHub, or is the mailing list better for that?

hnpl, 2024-09-24:
To create a new delay model, there are 3 steps,
1. Generating data points with the new PDK
2. Joining the data points with the op models
3. Adding the new delay model

Step 1. To generate the data points, you'll need to add a new xls_model_generation target to this build file, https://github.com/google/xls/blob/3ca5faad624ad55688b4957e2ddd3f7cdac2b753/xls/estimators/delay_model/BUILD#L266-L270.
To build the target, you can run,

```sh
bazel run //xls/estimators/delay_model:data_points-{pdk_name}
```

Step 2. To join the data points with the delay model's op models, you'll need to build the estimator_model_join target here https://github.com/google/xls/blob/3ca5faad624ad55688b4957e2ddd3f7cdac2b753/xls/estimators/BUILD#L95-L109 using the following command,

```sh
bazel build //xls/estimators:estimator_model_join
```

Then join the data points and the op models,

```sh
<path_to_estimator_model_join> --metric DELAY_METRIC --data_points data_points-{pdk_name}.textproto --op_models xls/estimators/delay_model/op_models.textproto --output {pdk_name}.textproto
```

Step 3. To add the new delay model, first, you need to copy the {pdk_name}.textproto file to xls/estimators/delay_model/models/. Then, you'll need to add the new target for the delay model like this one here, 
https://github.com/google/xls/blob/3ca5faad624ad55688b4957e2ddd3f7cdac2b753/xls/estimators/delay_model/models/BUILD#L48-L53

selimsandal, 2024-09-26:
@hnpl Thank you for the explanation. Before trying to generate a delay model 2 things I want to ask:

- The PDK I am targeting works with Synopsys tools and I am not familiar with Yosys enough to work with that PDK in Yosys, I am guessing if I edit the [run_op_characterization.py](https://github.com/google/xls/blob/main/xls/estimators/run_op_characterization.py) file to work with Design Compiler and PrimeTime I can get similar data. Will that work?

- While I am targeting an educational PDK I am testing my design on an Xilinx Ultrascale FPGA, I am guessing I need to schedule for that FPGA instead of the PDK to get usable clock speeds. I see that yosys has experimental support for synthesizing for similar FPGAs. Can I do that with Yosys / Vivado tools?

hnpl, 2024-09-26:
I moved the discussion here, https://github.com/google/xls/discussions/1634.

cdleary, 2024-10-06:
@selimsandal Assuming we can close this out since the original issues sounds like it was resolved, reopen if not, thanks!

closed 2024-10-06 by cdleary