google/xls #1626: [enhancement] XLS playground colab should be documented in our markdown docs

cdleary, 2024-09-24:
### What's hard to do? (limit 100 words)

Hard to remember the flags for the DSLX magic and how they may differ from the command line flags.

### Current best alternative workaround (limit 100 words)

Trying flags from the command line tools to see if they're the same.

### Your view of the "best case XLS enhancement" (limit 100 words)

We document the DSLX magic so it's easy to find the flags by searching the docs.

Ideally the DSLX magic could also be defined in the main repo per #1451 but that's nice-to-have.

cc @proppy

cdleary, 2024-09-24:
Note we also tried passing `--help` to the `%%dslx` magic but that didn't spit out help info.

proppy, 2024-09-24:
> Ideally the DSLX magic could also be defined in the main repo per #1451 but that's nice-to-have.

+1, I do have it there: https://github.com/google/xls/compare/main...proppy:xls:release but I need to send it as a proper PR.