google/xls #1633: Wrapped lines in struct construction should be indented

meheff, 2024-09-25:
Sample input with a reasonable formatting:

```
struct X { xxxxxxxxxxxxxxxxxxx: bool, yyyyyyyyyyyyyyyyyyy: u32 }

const aaaaaaaaaaaaaaaaaaaaaaaaa = u32:0;
const bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb = u32:0;
const ccccccccccccccccccccccccc = bool:false;

fn f() -> X {
    let a = X {
        xxxxxxxxxxxxxxxxxxx:
            if ccccccccccccccccccccccccc {
                aaaaaaaaaaaaaaaaaaaaaaaaa
            } else {
                bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
            },
        yyyyyyyyyyyyyyyyyyy: u32:0,
    };
    X {
        xxxxxxxxxxxxxxxxxxx:
            aaaaaaaaaaaaaaaaaaaaaaaaa == bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb,
        yyyyyyyyyyyyyyyyyyy: u32:0,
    }
}

```

Note that the wrapped lines in the struct construction are indented such that the field names stand out. Here's the dslx_fmt output:

```
struct X { xxxxxxxxxxxxxxxxxxx: bool, yyyyyyyyyyyyyyyyyyy: u32 }

const aaaaaaaaaaaaaaaaaaaaaaaaa = u32:0;
const bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb = u32:0;
const ccccccccccccccccccccccccc = bool:false;

fn f() -> X {
    let a = X {
        xxxxxxxxxxxxxxxxxxx: if ccccccccccccccccccccccccc {
            aaaaaaaaaaaaaaaaaaaaaaaaa
        } else {
            bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
        },
        yyyyyyyyyyyyyyyyyyy: u32:0,
    };
    X {
        xxxxxxxxxxxxxxxxxxx: aaaaaaaaaaaaaaaaaaaaaaaaa ==
        bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb,
        yyyyyyyyyyyyyyyyyyy: u32:0,
    }
}
```

cdleary, 2024-10-05:
Note that the current formatting for the `xxxxxxxxxxxxxxxxxxx` field in the first instance is intentional -- it always produces fewer lines and at a more outdented level, so should strictly save on lines.

The second instance's `xxxxxxxxxxxxxxxxxxx` field expression isn't formatted right, though, since the expression doesn't fit on the line it should be nested.

meheff, 2024-10-07:
> Note that the current formatting for the `xxxxxxxxxxxxxxxxxxx` field in the first instance is intentional -- it always produces fewer lines and at a more outdented level, so should strictly save on lines.

The `if`  could be in the same line as the field name. The significant bit is the indenting of the following lines so that the field names all stand out. I think this indenting should happen even if it results in more lines. basically field initialization `xxx: ....` should behave like `let xxxxx = ...` where the expression is indented if it is multiple lines.

closed 2024-10-14 by copybara-service[bot]