google/xls #1636: [fuzzer 2024-09-26_ac56] Different results between IR and IR-opt

hzeller, 2024-09-27:
`crasher_2024-09-26_ac56.x` reports a different result when evaluated before and after optimization

```
args: bits[21]:0x8_1001; bits[5]:0x1f; bits[37]:0x1f_ffff_ffff
evaluated opt IR (JIT), evaluated opt IR (interpreter), evaluated unopt IR (interpreter), interpreted DSLX, simulated =
   (bits[49]:0xffff_ffff_ffff, bits[1]:0x1, bits[1]:0x1, (bits[1]:0x1, bits[19]:0x5_959f))
evaluated unopt IR (JIT) =
   (bits[49]:0xffff_ffff_ffff, bits[1]:0x1, bits[1]:0x1, (bits[1]:0x1, bits[19]:0x3_2b3e))
```

To reproduce: Create a directory to collect all artifacts

```bash
mkdir /tmp/fuzzout
bazel build -c opt xls/fuzzer:run_crasher
bazel-bin/xls/fuzzer/run_crasher --run_dir=/tmp/fuzzout xls/fuzzer/crashers/crasher_2024-09-26_ac56.x
```

hzeller, 2024-09-27:
adding 'optimizer' label for now as this might be an optimizer issue.

ericastor, 2024-09-27:
Looks more likely to be an LLVM problem? Seems like a case where the optimizations are hiding an difference between the interpreter and the JIT.

allight, 2026-11-10:
This crasher (crasher_2024-09-26_ac56.x) is no longer in the known-failing lists of //xls/fuzzer:regression_tests, so it appears to be fixed (re-enabled in https://github.com/google/xls/commit/793d3dea05a26072facee992d0151f1a0c81d943).

closed 2026-11-10 by allight