google/xls #1653: [enhancement] Add runtime-constant values & channels

ericastor, 2024-10-08:
### What's hard to do? (limit 100 words)

Accept values from "configuration registers" and reference them in your circuit. These are values that can be assumed not to change between resets.

### Current best alternative workaround (limit 100 words)

In DSLX: Nothing great. The values can be sent through a normal channel, with the other end wired to always be ready and expose the value inside the configuration register on the data line. However, this makes the DSLX block unnecessarily complex.

In XLS IR: We can use single-value channels, but their behavior is underspecified & often implies aspects of state that we don't want.

### Your view of the "best case XLS enhancement" (limit 100 words)

Add a new type of channel that passes along values that do not change between resets ("param" channels? "const" channels?). This would need to be exposed in DSLX via some way to express that the value is runtime-constant, but not known at compile time.

It should also be possible to forward these values - or to send derived values - along this same type of runtime-constant channel between XLS procs.

ericastor, 2024-10-08:
@meheff - this is the issue we discussed today!

Inory-Megumi, 2026-09-24:
Any update for this issue