google/xls #1659: multi-dimensional chan array should use row major layout.

proppy, 2024-10-11:
**Describe the bug**

It seems that an multi-dimensional array of size `[M][N]` has to be indexed w/ indices in the following range `[0-N][0-M]` (column major) while intuitively one would assume `[0-M][0-N]` (row major) convention.

**To Reproduce**

```
proc SomeProc {
    some_chan_2d_array_r: chan<u32>[4][8] in;

    config() {
      let (some_chan_2d_array_s, some_chan_2d_array_r) = chan<u32>[4][8]("some_chan_2d_array");
      (some_chan_2d_array_r,)
    }

    init {  }

    next(state: ()) {
        let (tok, _) = recv(token(), some_chan_2d_array_r[0][7]);
    }
}
```
produces the following error:
```
TypeInferenceError: chan(uN[32], dir=in)[4] Index has a compile-time constant value 7 that is out of bounds of the array type.
```
meansing that `some_chan_2d_array_r[0]` is of type `chan(uN[32], dir=in)[4]` instead of being `chan(uN[32], dir=in)[8]`.

**Expected behavior**

Indexing works like cpp semantic: https://en.cppreference.com/w/cpp/language/array#Multidimensional_arrays

**Additional context**

See: https://colab.research.google.com/gist/proppy/42d3349617c7436e7e802d54ae101c64/xls-playground-sandbox.ipynb

proppy, 2024-10-11:
the behavior is at least coherent with regular array:
```
0003: fn muladd(a: u8, b: u8) -> u8 {
0004:     let arr = u8[2][4]:[[u8:1, u8:2, u8:3, u8:4], [u8:5, u8:6, u8:7, u8:8]];
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^--------------------------------------------------^ XlsTypeError: Annotated array size 4 does not match inferred array size 2.
```

mikex-oss, 2024-10-11:
This is just https://github.com/google/xls/issues/616 right? Seems WAI unless we change arrays in general.

proppy, 2024-10-11:
Ype, thanks for the ref https://github.com/google/xls/issues/616#issuecomment-2286766187 I thought it was specific to the newly array of chans first, but that's just how multi-dimentional array currently work, https://github.com/google/xls/issues/616#issuecomment-2286766187 comments describe it better than this bug so closing.

closed 2024-10-11 by proppy