google/xls #1660: LSP: highlight all other places of a symbol under the cursor (`textDocument/documentHighlight`)

hzeller, 2024-10-11:
When the cursor is over a symbol, highlight all other places where this particular symbol is used or where it is defined.

This requires to understand the scope of the symbol, but that information should already be fully there (the rename functionality will already do that).

Spec:
https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_documentHighlight

This need to receive `verible::lsp::DocumentHighlightParams` and return a `std::vector<verible::lsp::DocumentHighlight>` which are already defined in the `lsp-protocol.yaml`.
The LSP will then advertise that it implements a `documentHighlightProvider` in its capabilities.

hzeller, 2024-10-11:
@cdleary this sounds like a good recreational LSP feature implementation fun for you.

cdleary, 2024-10-13:
@hzeller Sure does!

Happy to knock this out.

Side note: one of the bigger things in terms of day to day annoying-ness are around the buffer management with the language server are when there are transitive dependencies for modules and you make a fix in a module your "outer" module depends upon. I know you had some solution for this, haven't had time to scope it out, LMK if you have pointers for me or if it's a simple fix for you to take on.

PR #1661 (2024-10-14): [dslx:ls] Add support for DocumentHighlight capability.
Fixes #1660

Sample view when some logging is added to confirm it's the feature is being triggered appropriately:

[image]
closed 2024-10-14 by copybara-service[bot]