google/xls #1668: map w/ parametric builtin doesn't convert

proppy, 2024-10-22:
**Describe the bug**

trying to convert a map that uses a parametric builtin fails with the following error:
```
```INTERNAL: Invalid builtin name for map: widening_cast
```

**To Reproduce**
```
fn f() -> u8[4] {
  map(zero!<u7[4]>(), widening_cast<u8>)
}
```

richmckeever, 2024-10-23:
I think the underlying issue here is that type parametrics (as opposed to u32) are only supported for certain builtins, and use hacks that are hard to propagate to a map call. A ParametricEnv can't even capture type parametrics, partly because an InterpValue can't be a type. widening_cast is not even supported by the bytecode interpreter, so in some contexts that would fail with a different error. The work on unifying type inference should clean this up.

mikex-oss, 2025-09-10:
This no longer errors in TIv2 at build time, but as mentioned in the above comment, it still seems to fail in the bytecode interpreter at test runtime.

`: internal error: UNIMPLEMENTED: BytecodeInterpreter: builtin function "widening_cast" not yet implemented.`