google/xls #1677: Bad error message for incorrect sizeof usage

meheff, 2024-10-30:
foo.x
```
pub type foo_t = u32;
```

main.x
```
import std;
import foo;

fn f() -> () {
  let s: u32 = std::sizeof(foo::foo_t);
}
```

Repro:
```
$ interpreter_main bad.x --dslx_path .
E1030 13:03:32.587061  991027 deduce_invocation.cc:237] INTERNAL: XLS_RET_CHECK failure (third_party/xls/dslx/type_system/deduce_invocation.cc:237) !type->IsMeta() parametric arg: foo::foo_t type: typeof(uN[32])
=== Source Location Trace: ===
third_party/xls/common/status/status_builder.cc:197

0x5601c71a5cc9: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x5601c4049097: xls::dslx::AppendArgsForInstantiation()
0x5601c404957f: xls::dslx::DeduceInvocation()
0x5601c402cc7e: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
...

```

richmckeever, 2024-10-30:
The error at hand probably needs to be fixed for other scenarios, but maybe we should turn sizeof into a builtin and make it work for anything with an exposable size.

meheff, 2024-10-30:
+1 to allowing `std::sizeof(Type)`.

richmckeever, 2026-10-27:
The error message now is better than it used to be: "Cannot pass a type as a function argument", but it could be more explanatory, e.g. "Did you mean `bit_count<foo::foo_t>()`?"

axman6, 2026-11-04:
Taking a look at this, my first thought was that maybe there should be a `std::sizeof_type<T>() { sizeof(zeros!<T>()) }` but I guess that's already what `bit_count` does. I'll see if I can improve the special case error for `sizeof(Type)`.