google/xls #1679: Parametric function arg type inference fails due to nested unresolved type

mikex-oss, 2024-10-30:
**Describe the bug**
It feels like type inference goes "too far" and generates a type inference error when the arg is a parametric struct, because the struct member can't be fully resolved, even though the parametric function is fully specified from the struct type.

```
0009:     foo(s);
~~~~~~~~~~~~~~^ XlsTypeError: Mismatch between parameter and argument types (after instantiation).
Type mismatch:
   SomeParamStruct { x: uN[width(T)] }
vs SomeParamStruct { x: uN[3] }
```

**To Reproduce**
```
struct SomeParamStruct<T: u32> { x: bits[std::clog2(T)] }

fn foo<T: u32>(s: SomeParamStruct<T>) -> bits[T] { s.x }

fn main() {
    let s = SomeParamStruct<u32:8> { x: u3:5 };
    foo(s);
}
```

**Expected behavior**
Since the specialization of `foo` is known to be for `SomeParamStruct<u32:8>` it feels like this should work. You get a similar error if you don't specify the parametrics for `s`, but that makes more sense since you would need to back out the `clog2` for the type of member `x`.

mikex-oss, 2024-11-01:
Note that it doesn't help to move the `clog2` out of the type and into a separate default parameteric:

```
struct SomeParamStruct<T: u32, SZ_T: u32 = {std::clog2(T)}> { x: bits[SZ_T] }

fn foo<T: u32>(s: SomeParamStruct<T>) -> bits[T] { s.x }

fn main() {
    let s = SomeParamStruct<u32:8> { x: u3:5 };
    foo(s);
}
```

produces the same error.

richmckeever, 2026-10-27:
This is not an issue anymore, provided you correctly specify the return type of foo as `bits[std::clog2(T)]`. With the code exactly as originally posted, you would now get an error that `s.x` doesn't match the function's declared return type, which is true.

closed 2026-10-27 by richmckeever