google/xls #1683: [enhancement] DSLX support for option types

grebe, 2024-10-31:
### What's hard to do? (limit 100 words)

Until general polymorphism (see #1646) is supported, we don't have a good way to write our own option type. It's also hard to write polymorphic functions that would use such a type (e.g. a `value_or()`).

### Current best alternative workaround (limit 100 words)

Write an option type for each type you want to encapsulate, or just include a `valid` entry in a struct.

### Your view of the "best case XLS enhancement" (limit 100 words)

#618 and #1646 is the ultimate goal. It might be worth:

- using option as a motivating example on the path towards those goals
- building not-general-purpose facilities to support option type before fully supporting polymorphism

grebe, 2024-11-14:
Yeah, I'm suggesting we do that unless we think it's not much more work to do something more generic.

mikex-oss, 2025-05-06:
I ran into another cumbersome scenario in DSLX that this might address.

Namely, detecting if any non-default match arm was taken. Since we don't have mutable variables, you can't do something like:

```
let matched: bool = true;
let result = match x {
        FOO => u32:23,
        BAR => u32:42,
        ...
        _ => {
            matched = false;
            u32:0
        },
};
```

Instead, you need to supply the true value to every match arm. 

let (result, matched) = match x {
        FOO => (u32:23, true),
        BAR => (u32:42, true),
        ...
        _ => (u32:0, false),
};
```

When the number of match arms is large (and the default covers an equally large space), this can be quite tedious. I had some success with an LLM to do this change, but that can be finicky. 

You could envision this `Option` type working like in Rust:

```
let result: Option<u32> = match x {
        FOO => u32:23,
        BAR => u32:42,
        ...
        _ => None,
};
let matched = result.is_some();
```