google/xls #1708: `unroll_for!` is not documented

dplassgit, 2024-11-12:
It's not documented in https://google.github.io/xls/dslx_reference/ or https://google.github.io/xls/dslx_std/ or https://google.github.io/xls/tutorials/crc32/ (which talks about for loops)

Svastikkka, 2026-10-31:
Hi, I'd like to work on this.

Since this was filed, DSLX has added const for, and in the parser it's the same construct as unroll_for! (both are ConstFor, only the keyword differs). xls/examples/const_for.x uses the new form, and neither is in the reference yet.

My plan is to add a section after the for expression section in dslx_reference.md. It would cover the syntax, the fact that the loop is unrolled at compile time so the range has to be constant, and the proc use case (spawning procs over channel arrays, as in the procs tutorial), with a short example.

Should the section be written for const for, with unroll_for! mentioned as the older spelling, or the other way round?

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `unroll_for!` is now documented in `docs_src/dslx_reference.md`, `docs_src/dslx_std.md`, and `docs_src/tutorials/how_to_use_procs.md`. (Closed by Gemini)

closed 2026-11-14 by allight