google/xls #1710: [enhancement:DSLX] Validate "input" as a parameter at the DSLX level (long before codegen)

cdleary, 2024-11-13:
### What's hard to do? (limit 100 words)

Right now if you use "input" as a parameter name for a function, DSLX will accept it happily, but if you try to run that through code generation it will give you:

```
Error: INVALID_ARGUMENT: Port `input` is a keyword.; Running pass #15: Legalize names to avoid keywords [short: name_legalization]; Running pass #15: Top level codegen pass pipeline [short: codegen]
```

### Current best alternative workaround (limit 100 words)

Rename the argument by hand.

### Your view of the "best case XLS enhancement" (limit 100 words)

Flag this as either a warning or an error at the DSLX level. I'd welcome some input on whether it's best to be a warning or an error.

cdleary, 2024-11-13:
Also note that `in` is a DSLX keyword, so generically describing something that is "an input of some kind" can cause a user to bump into a few things.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `PreTypecheckPass::HandleParam` in `xls/dslx/frontend/semantics_analysis.cc:430-444` checks function and proc parameter identifiers against `kSystemVerilogKeywords` and emits a `WarningKind::kVerilogKeywordName` warning when a parameter is named `input` or another reserved Verilog keyword. (Closed by Gemini)

closed 2026-11-14 by allight