google/xls #1714: [enhancement] variant_count!<> builtin

grebe, 2024-11-14:
### What's hard to do? (limit 100 words)

Similar to `u32::MAX`, it would be nice to have a way to get the number of elements in an enum. Similar to rust's [std::mem::variant_count](https://doc.rust-lang.org/std/mem/fn.variant_count.html).

### Current best alternative workaround (limit 100 words)

Make a const yourself and hope you keep it up to date. 

### Your view of the "best case XLS enhancement" (limit 100 words)

A builtin `std::variant_count!<>` or a colon-ref (like `u32::MAX`), e.g. EnumType::NUM_VARIANTS.

mikex-oss, 2024-11-14:
Surprisingly, it seems Rust does not yet have a robust story for this. `variant_count` is still not stabilized and the latest comments in the issue seem to indicate going in a different direction with magic constants like  `MyEnum::enum#VARIANT_COUNT` .

cdleary, 2024-11-14:
I think this also gets more urgent if we had iota-assigning enums -- right now you have to name all the values by hand so making `COUNT` with the underlying is comparatively not so burdensome. I'm not sure if we have an issue for that? We do have #1306 for getting proper sum types.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): DSLX added the `element_count<T>()` builtin (`xls/dslx/frontend/builtin_stubs.x:64`, `xls/dslx/dslx_builtins.h:51`, implemented in `xls/dslx/type_system/deduce_utils.cc:551-570` and `xls/dslx/type_system_v2/inference_table_converter_impl.cc`), which returns the number of variants for an `EnumDef` (`enum_type.values().size()`) as tested in `xls/dslx/type_system/typecheck_module_test.cc:4638` (`element_count<A>()`). (Closed by Gemini)

closed 2026-11-14 by allight