google/xls #1719: dslx_fmt strips inline struct instantiation comments

mikex-oss, 2024-11-15:
**Describe the bug**
Similar to #1260, except for struct instantiations. You can't comment on a member initialization as the formatter removes it.

**To Reproduce**
```
struct Foo {
    // this is a comment
    a: bits[1],
    // this is another comment
    b: bits[1],
}

fn main() {
    let _foo = Foo { 
        // Because blah
        a: u1:0, 
        // And blah
        b: u1:1 
    };
}
```

is reformatted to:

```
struct Foo {
    // this is a comment
    a: bits[1],
    // this is another comment
    b: bits[1],
}

fn main() {
    let _foo = Foo { a: u1:0, b: u1:1 };
}
```

**Expected behavior**
Comments should stay anchored where they are.

dplassgit, 2024-11-15:
Yup, pretty much the same bug as issue #1697

dplassgit, 2024-12-16:
This is *detected* as of commit 2398f4b but then the file can't be formatted.

closed 2024-12-19 by copybara-service[bot]