google/xls #1721: unroll_for does not unroll constants

mikex-oss, 2024-11-16:
**Describe the bug**
Constants can not be declared in unroll_for loops, even though they work with `let` binding or used inline.

TypeMissingError: node: 0x30d27fe5f6c0 user: 0x30d27e4e4d98 internal error: AST node is missing a corresponding type: B 0x30d27fe5f6c0 (kind: NameDef) defined @ foo.x10:15-10:16. This may be due to recursion, which is not supported.

**To Reproduce**
```
fn main() -> bool {
    unroll_for! (i, b): (u32, bool) in u32:0..u32:10 {
        const B: bool = i == u32:1;
        b ^ B
    }(true)
}
```

The following work:

```
fn main() -> bool {
    unroll_for! (i, b): (u32, bool) in u32:0..u32:10 {
        let B: bool = i == u32:1;
        b ^ B
    }(true)
}
```

```
fn main() -> bool {
    unroll_for! (i, b): (u32, bool) in u32:0..u32:10 {
        b ^ (i == u32:1)
    }(true)
}
```

**Expected behavior**
Since the loop index is constexpr, it seems reasonable to define a `const` that depends on it. I would expect it to work like variables.

**Additional context**
Above is for minimal repro, but a more realistic example would be where `B` is defined for multiple parametric function invocations.

```
fn foo<B: bool>(x: u32) -> u32 { if B { x } else { u32:0 } }

fn bar<B: bool>(x: u32) -> bool { if B { x != u32:0 } else { x != u32:1 } }

fn main(x: u32) {
    unroll_for! (i, _): (u32, ()) in u32:0..u32:10 {
        const B = (i % u32:2 == u32:1);
        foo<B>(x);
        bar<B>(x);
    }(());
}
```

Mars-Cat2023, 2025-01-21:
Hello.
May I ask do you have any follow-ups for this?

I also encounter the issues that I cannot declare any constants by "const" or by "let".

```
proc matmul_scale <ROWS: u32, COLS: u32, W_ROWS: u32, A_ROWS: u32, A_COLS: u32> {

    let W_COLS = A_ROWS;
    let Z_ROWS = W_ROWS;
    let Z_COLS = A_COLS;
    
    config(...) {...}
    init { () }
    next(state: ()) {
    ...
}
```

I will get this error:
```
0194:                  let W_COLS = A_ROWS;
 ParseError: Unexpected token in proc body: keyword:let; want one of 'config', 'next', or 'init'

```

It doesn't work for "const" either.
```
0191:     const W_COLS:u32 = A_ROWS;
TypeMissingError: node: 0x55d6633d5150 user: 0x55d6633d5dd0 internal error: AST node is missing a corresponding type: A_ROWS 0x55d6633d5150 (kind: NameDef) defined @ E5_E4M3_matrixmul_896.x:186:55-186:61. This may be due to recursion, which is not supported.
```

mikex-oss, 2025-01-21:
@Mars-Cat2023 sounds like you're hitting unrelated issues. Please file a separate bug, as I don't see any unroll_for in your code.

That being said, `let` in proc scope will presumably always be illegal, similar to how you can't create a `let` binding in top-level scope. Also, proc-level consts are tracked in https://github.com/google/xls/issues/1580.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): In `TypeSystemV2`, `ConstantCollector::HandleUnrollFor` (`xls/dslx/type_system_v2/constant_collector.cc:516-605`) unrolls `unroll_for!` loops into a `StatementBlock` with shadowed constant `Let` bindings and runs `CreatePopulateTableVisitor` and `ConvertSubtree` over the unrolled statements, allowing `const` declarations inside `unroll_for!` bodies to depend on the loop index `i`. (Closed by Gemini)

closed 2026-11-14 by allight