google/xls #1726: Array of channels causes internal error

m-torhan, 2024-11-18:
**Describe the bug**
Creating an array of channels and then using it in loop causes internal error.

**To Reproduce**
Steps to reproduce the behavior:
1. Write following code:
```rust
proc Foo {
    in_0_r: chan<u32> in;
    in_1_r: chan<u32> in;
    out_s: chan<u32> out;

    config(in_0_r: chan<u32> in, in_1_r: chan<u32> in, out_s: chan<u32> out) {
        (in_0_r, in_1_r, out_s)
    }

    init { u1:0 }
    
    next (state: u1) {
        let in_r = [
            in_0_r,
            in_1_r,
        ];
        let (tok, val) = recv(join(), in_r[state]);
        send(tok, out_s, val);

        state + u1:1
    }
}
```
2. Build Verilog target
```bazel
xls_dslx_library(
    name = "foo_dslx",
    srcs = ["foo.x"],
    deps = [],
)

xls_dslx_verilog(
    name = "foo_verilog",
    codegen_args = {
        "module_name": "Foo",
        "delay_model": "asap7",
        "pipeline_stages": "8",
        "reset": "rst",
        "use_system_verilog": "false",
    },
    dslx_top = "Foo",
    library = ":foo_dslx",
    verilog_file = "foo.v",
)
```
```bash
bazel build //:foo_verilog
```
3. Observe error
```none
ERROR: /mnt/hdd_0/Workspace/google-xls/BUILD:82:17: Converting DSLX file to XLS IR: foo.x failed: (Exit 1): ir_converter_main failed: error
 executing ConvertDSLX command (from target //:foo_verilog) bazel-out/k8-opt-exec-ST-13d3ddad9198/bin/xls/dslx/ir_convert/ir_converter_main '--dslx_path=:${PWD}:bazel-out/k8-fastbuild/bin:bazel-out/k8-fastbuild/bin::bazel-out/k8-fastbuild/bin/' '--top=Foo' ... (remaining 5 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
E1118 12:28:39.701892       2 function_converter.cc:556] INTERNAL: XLS_RET_CHECK failure (xls/dslx/ir_convert/function_converter.cc:556) std::holds_alternative<CValue>(ir_value)
0x5a9fdd9691fa: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x5a9fdd60e619: absl::lts_20240722::StatusOr<>::StatusOr<>()
0x5a9fdd60f0ea: xls::dslx::FunctionConverter::Use()
0x5a9fdd622dcc: xls::dslx::FunctionConverter::HandleArray()
0x5a9fdd640752: xls::dslx::FunctionConverterVisitor::HandleArray()
0x5a9fdd88489b: xls::dslx::Array::Accept()
0x5a9fdd60c765: xls::dslx::FunctionConverterVisitor::Visit()
0x5a9fdd613f12: xls::dslx::FunctionConverter::HandleLet()
0x5a9fdd6409a2: xls::dslx::FunctionConverterVisitor::HandleLet()
0x5a9fdd886cbb: xls::dslx::Let::Accept()
0x5a9fdd60c765: xls::dslx::FunctionConverterVisitor::Visit()
0x5a9fdd649555: std::__1::__variant_detail::__visitation::__base::__dispatcher<>::__dispatch[abi:ue170006]<>()
0x5a9fdd63e83d: xls::dslx::FunctionConverter::HandleStatement()
0x5a9fdd6404f2: xls::dslx::FunctionConverterVisitor::HandleStatement()
0x5a9fdd886b1b: xls::dslx::Statement::Accept()
0x5a9fdd60c765: xls::dslx::FunctionConverterVisitor::Visit()
0x5a9fdd63e4a3: xls::dslx::FunctionConverter::HandleStatementBlock()
0x5a9fdd640832: xls::dslx::FunctionConverterVisitor::HandleStatementBlock()
0x5a9fdd8863bb: xls::dslx::StatementBlock::Accept()
0x5a9fdd60c765: xls::dslx::FunctionConverterVisitor::Visit()
0x5a9fdd6399df: xls::dslx::FunctionConverter::HandleProcNextFunction()
0x5a9fdd5f1965: xls::dslx::(anonymous namespace)::ConvertCallGraph()
0x5a9fdd5f395c: xls::dslx::ConvertOneFunctionIntoPackageInternal<>()
0x5a9fdd5f34a5: xls::dslx::ConvertOneFunctionIntoPackage()
0x5a9fdd5f4a62: xls::dslx::ConvertFilesToPackage()
0x5a9fdd5a82c0: main
0x7245c9763e08: [unknown]
0x7245c9763ecc: __libc_start_main
0x5a9fdd5a7c15: _start

Error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/ir_convert/function_converter.cc:556) std::holds_alternative<CValue>(ir_value) Target //:f
oo_verilog failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 0.527s, Critical Path: 0.31s
INFO: 3 processes: 2 internal, 1 linux-sandbox.
ERROR: Build did NOT complete successfully
```

**Expected behavior**
The target is built with no errors.

**Environment (this can be helpful for troubleshooting):**
 - OS: Arch Linux x86_64
 - Kernel version: 6.11.6-arch1-1
 - XLS hash: 11e5e7dd6453c23c01a86f591f6bf10043ca1f0c

richmckeever, 2024-11-18:
We should change this to return a better error, but channel arrays are intended to be used
- Without construction time aliasing (other than by forwarding from one proc config to another). In other words, a channel should be in an array from the beginning of its existence.
- With constexpr indices.

Both restrictions are violated by the example. Here's a modified version I would expect to work:

```
proc Foo {
    in_r: chan<u32>[2] in;
    out_s: chan<u32> out;

    config(in_r: chan<u32>[2] in, out_s: chan<u32> out) {
        (in_r, out_s)
    }

    init { u1:0 }
    
    next (state: u1) {
        let (tok, val) = if state == u32:0 { recv(join(), in_r[0]) } else { recv(join(), in_r[1]) };
        send(tok, out_s, val);
        state + u1:1
    }
}
```

If used in a loop, it needs to be an unroll_for! loop, which expands to a series of copies of the loop body in which the indices are constexpr.

m-torhan, 2024-11-19:
Thanks for clarification @richmckeever. I tried to build the modified version and I got another error:
```none
E1119 08:52:19.718691       3 z3_ir_translator.cc:1403] INTERNAL: XLS_RET_CHECK failure (xls/solvers/z3_ir_translator.cc:1403) ty->IsTuple()
0x60a1c46102da: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x60a1c3368ff9: absl::lts_20240722::StatusOr<>::StatusOr<>()
0x60a1c3373684: xls::solvers::z3::IrTranslator::FromLeafTypeTree()
0x60a1c3376152: xls::solvers::z3::IrTranslator::HandleSel()
0x60a1c44c7125: xls::Node::VisitSingleNode()
0x60a1c44c807a: xls::Node::Accept()
0x60a1c44bcfbf: xls::FunctionBase::Accept()
0x60a1c336899e: xls::solvers::z3::IrTranslator::CreateAndTranslate()
0x60a1c2e52dcb: xls::ComputeMutualExclusion()
0x60a1c2e55fb4: xls::MutualExclusionPass::RunOnFunctionBaseInternal()
0x60a1c4091a72: xls::SchedulingOptimizationFunctionBasePass::RunInternal()
0x60a1c2e4c5ca: xls::PassBase<>::Run()
0x60a1c2e4d730: xls::CompoundPassBase<>::RunNested()
0x60a1c2e4cdf3: xls::CompoundPassBase<>::RunInternal()
0x60a1c2e4c5ca: xls::PassBase<>::Run()
0x60a1c2df1baa: xls::Schedule()
0x60a1c2df4f57: xls::ScheduleAndCodegen()
0x60a1c2da36e1: xls::(anonymous namespace)::RealMain()
0x60a1c2da3287: main
0x7482e5f4de08: [unknown]
0x7482e5f4decc: __libc_start_main
0x60a1c2da3165: _start
```
There might be something wrong with using channel arrays inside `if` statements. When I replaced `if` with `recv_if`s, the code was built with no errors.
Modified `next`:
```rust
next (state: u1) {
    let (tok_0, val_0) = recv_if(join(), in_r[0], state == u1:0, u32:0);
    let (tok_1, val_1) = recv_if(join(), in_r[1], state == u1:1, u32:0);
    let tok = join(tok_0, tok_1);
    let val = if state == u1:0 { val_0 } else { val_1 };
    send(tok, out_s, val);
    state + u1:1
}
```