google/xls #1727: Seemingly valid bit vector indexing yields parse errors

meheff, 2024-11-19:
Code is below. Errors and the problematic code are in the comments

```rust
# my_lib
pub const W = s32:16;
pub const Z = s32:24;
```

```rust
import my_lib;

pub const A = s32:16;
pub const B = s32:24;

fn foo(x: u32) -> () {
// 0013:   let a = x[my_lib::W : s32:24];
// ~~~~~~~~~~~~~~~~~^------------------^ TypeInferenceError: uN[32] Value to index is not an array.
//  let a = x[my_lib::W : s32:24];

// 0017:   let b = x[my_lib::W : my_lib::Z];
// ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^ ParseError: Old-style cast only permitted for constant arrays/tuples and literal numbers.
//  let b = x[my_lib::W : my_lib::Z];

  // OK.
  let c = x[A:my_lib::Z];

  // OK.
  let d = x[A:B];

  ()
}
```

mikex-oss, 2024-11-19:
Agreed this is confusing, but I guess this is because indices should be raw decimal literals, while constants always have a type?

Sized literals are not ever shown in the documentation:

https://google.github.io/xls/dslx_reference/#bit-slice-examples
https://github.com/google/xls/blob/main/xls/dslx/tests/bit_slice_syntax.x

meheff, 2024-11-19:
> Agreed this is confusing, but I guess this is because indices should be raw decimal literals, while constants always have a type?
>
> Sized literals are not ever shown in the documentation: [...]

Indices using named constants works fine if the constants are locally declared, but not if they are imported (well, at least if the start index is imported). So support is inconsistent.

mikex-oss, 2024-12-20:
Looks like it's not just imported constants, but locally defined constants which are expressions referencing an imported constant.