google/xls #1731: LSP: `DSLX_PATH` stopped working

hzeller, 2024-11-19:
The `DSLX_PATH` environment variable used to be the way to tell he language server what the base path is for finding import paths.

Let's assume the XLS checkout is in a subdirectory of my homedir: `/home/myhomedir/src/xls` and thelanguage server configured for the editor.

Now opening a DSLX file. To make sure the language server finds all import paths, I am setting the `DSLX_PATH` to the root of the directory:

```
DSLX_PATH=/home/myhomedir/src/xls emacs src/xls/xls/dslx/stdlib/float32.x
```

This now lets the language server crash while opening the file:
```
XLS testing language server
Path configuration:
        stdlib=xls/dslx/stdlib
        dslx_path=/home/myhomedir/src/xls
        cwd=/home/myhomedir/src/xls
F1119 11:00:35.471252 1009793 language_server_adapter.cc:232] Check failed: absl::StartsWith(importer_filename, "file://") || absl::StartsWith(importer_filename, "memfile://") importer_filename: xls/dslx/stdlib/apfloat.x imported: "xls/dslx/stdlib/std.x"
```

It looks like some part of the code assumes regular host filenames, while others assume the `file://` URI notation, which then clashes here.

hzeller, 2024-11-19:
Looks like this broke with the introduction of the virtual filesystem in 1b175c677d758252af957a6c0cbba675cdda9b46
Maybe @cdleary sees immediately what is amiss.

cdleary, 2024-11-28:
@cdleary Please close if this is working for you now, I think with the test I added it should be 🙏

hzeller, 2024-12-01:
Yes, works now, thanks!

closed 2024-12-02 by hzeller