google/xls #1735: dslx-fmt::off in for loop header deletes the parent proc

mikex-oss, 2024-11-20:
**Describe the bug**
I tried to disable formatting of a long for loop accumulator type annotation, due to https://github.com/google/xls/issues/1685 making the loop hard to read.

It deleted my entire proc.

**To Reproduce**
Run the following through the formatter:

```
struct FooType { a: u32, b: u32 }

struct BarType { c: u32, d: u32 }

const NUM_ELEMS = u32:8;
const NUM_BLOCKS = u32:2;

proc A {
    config() {  }

    init {  }

    next(_: ()) {
        // some comment
        let _some_import_code_here = true;

        let (_foo, _bar, _baz) =
            for (i, (foo, bar, baz)): (
                // dslx-fmt::off
                u32, (FooType[NUM_ELEMS][NUM_BLOCKS], BarType[NUM_ELEMS][NUM_BLOCKS], 
                      bool[NUM_ELEMS][NUM_BLOCKS])
                ) in range(u32:0, 8) {
                // dslx-fmt::on
                
                // this is another cool comment
                (foo, bar, baz)
            }((zero!<FooType[8][2]>(), zero!<BarType[8][2]>(), all_ones!<bool[8][2]>()));

        // the end
        let _my_grand_finale_here = true;
    }
}
```

Output:

```
struct FooType { a: u32, b: u32 }

struct BarType { c: u32, d: u32 }

const NUM_ELEMS = u32:8;
const NUM_BLOCKS = u32:2;

// some comment

// dslx-fmt::off
```

**Expected behavior**
Source outside the disabled formatting section should be untouched. All the code should still be present.

dplassgit, 2024-11-21:
This will no longer be allowed (and the user will be notified!)

Instead, use: 

```
let (_foo, _bar, _baz) =
            for (i, (foo, bar, baz)): (
                // dslx-fmt::off
                u32, (FooType[NUM_ELEMS][NUM_BLOCKS], BarType[NUM_ELEMS][NUM_BLOCKS], 
                      bool[NUM_ELEMS][NUM_BLOCKS])
                )
                // dslx-fmt::on
                 in range(u32:0, 8) {
                
                // this is another cool comment
                (foo, bar, baz)
            }((zero!<FooType[8][2]>(), zero!<BarType[8][2]>(), all_ones!<bool[8][2]>()));
```

dplassgit, 2024-12-02:
Blocked by Issue #1756

dplassgit, 2024-12-04:
(Analysis copied from #1746:)

I think this happens because the AST contains two copies of everything from a proc: the "extracted" methods (e.g., matmul.init, matmul.config, matmul.next) and the "original" proc node. They're processed in sequence, and the format disabler is getting confused regarding which nodes should be converted to VerbatimNodes.

See https://github.com/google/xls/issues/1029