google/xls #1736: [cleanup] Should dependency header inclusions have `external/` prefix?

cdleary, 2024-11-20:
I think usually the fact that bazel homes workspace-pulled things in a directory named `external/` is an implementation detail and we add the C++ header locations to the include path. It looks like primarily z3 and verible are the ones with this property. Probably we just need to tweak the workspace a bit to clean this up if it's unintentional, but let me know if this is actually a recommended practice now?

https://github.com/search?q=repo%3Agoogle%2Fxls+%22external%2F%22+language%3Acpp&type=code

cdleary, 2024-11-20:
Sounds like it's just a copybara artifact, so I can look at cleaning it up.

hzeller, 2024-11-21:
I think it just resulted from verible/ having the 'interesting' header in 'common/' so without the prefix it would just look like we're including things from `common/lsp/foo.h` which sounded like 'common/' comes from XLS.
Similarly the z3/ headers looked very generic without (compare `external/z3/src/api/z3_api.h` with `src/api/z3_api.h`).

Since you added the `external/verible/` header (and it improved the readability) we've used that as well in z3 for the same reason.

So the improved readability is the main reason, but I don't think there is a strong recommendation either way.

meheff, 2024-11-21:
Can we use the http_archive attribute add_prefix to resolve this?

https://bazel.build/rules/lib/repo/http#http_archive-add_prefix

So our verible rule, for example, will look something like:

```
http_archive(
        name = "verible",
        sha256 = "946625a1527d0a97772ea031ab7358af29e61258c189a2ab0d9533b43e71f35b",
        strip_prefix = "verible-" + VERIBLE_TAG.lstrip("v"),
        urls = ["https://github.com/chipsalliance/verible/archive/refs/tags/" + VERIBLE_TAG + ".tar.gz"],
        add_prefix = "verible",
        ....
    )
```

Then I think we should be able to include headers like `#include "verible/common/...` and build deps become `@verible//verible/common/...`. There may be complications if some dep uses the same library as well. Then its paths need to be adjusted as well.

cdleary, 2024-11-29:
Closing as complete, C++ files don't have #includes of `external/` directories anymore

closed 2024-11-29 by cdleary