google/xls #1772: [enhancement]

rmlarsen, 2024-12-06:
### What's hard to do? (limit 100 words)

It would be nice to be able request a zero prefix for `trace_fmt` using the `:#x" format.

**To Reproduce**
```
let x = u20:0x00800;
trace_fmt!("{:#x}", x);
```

The program outputs `0x800`. 

It would be nice to be able to output it in the original format for compatibility with other tools. 

### Current best alternative workaround (limit 100 words)

None that I'm aware of.

### Your view of the "best case XLS enhancement" (limit 100 words)

The syntax `{:#0x}` would a natural extension, similar to the `{:0x}` format.

mikex-oss, 2024-12-13:
Seems reasonable but isn't the workaround:

```
let x = u20:0x00800;
trace_fmt!("0x{:0x}", x);
```

meheff, 2024-12-13:
Explicitly adding the prefix and using `0` is an easy workaround but we should probably support the combination of `#` and `0` though it's probably not high priority given the workaround.