google/xls #1776: [fuzz] Crash in LLVM via eval_ir_main, target-independent node not lowered

cdleary, 2024-12-08:
**Describe the bug**

IR is attached.

```[sample.ir.txt](https://github.com/user-attachments/files/18053493/sample.ir.txt)

✓ /tmp/crashers-2024-12-07/95cce86e  cdleary@vision 10:18:10
$ /home/cdleary/.cache/bazel/_bazel_cdleary/e3482f322694a8d1a6845914a1a3d787/execroot/com_google_xls/bazel-out/k8-dbg/bin/xls/tools/eval_ir_main --testvector_textproto=./testvector.pbtxt --use_llvm_jit ./sample.ir --logtostderr
t776: ch = <<Deleted Node!>>
This target-independent node should have been selected!
UNREACHABLE executed at external/llvm-project/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.cpp:1247!
Aborted
```

**Expected behavior**
No LLVM crash

**Environment (this can be helpful for troubleshooting):**
 - OS: Ubuntu 24.04.1 LTS

**Additional context**
Was fuzzing overnight after #1774

allight, 2024-12-19:
Odd configuring msan removes the error (though that does actually change the code the jit creates so its not impossible that would be expected). Running with asan points at LLVM X86DAGToDAGISel as the allocator of that node.

This could be an llvm issue.

```
=================================================================
==3371915==ERROR: AddressSanitizer: use-after-poison on address 0x7d4b45e5a71c at pc 0x562956e26749 bp 0x7ffd1a242da0 sp 0x7ffd1a242d98
READ of size 4 at 0x7d4b45e5a71c thread T0
    #0 0x562956e26748 in getIROrder third_party/llvm/llvm-project/llvm/include/llvm/CodeGen/SelectionDAGNodes.h:772:40
    #1 0x562956e26748 in (anonymous namespace)::RegReductionPQBase::getNodeOrdering(llvm::SUnit const*) const third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1804:27
    #2 0x562956e2c406 in (anonymous namespace)::src_ls_rr_sort::operator()(llvm::SUnit*, llvm::SUnit*) const third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:2658:26
    #3 0x562956e2c2e2 in popFromQueueImpl<(anonymous namespace)::src_ls_rr_sort> third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1853:9
    #4 0x562956e2c2e2 in llvm::SUnit* (anonymous namespace)::popFromQueue<(anonymous namespace)::src_ls_rr_sort>(std::__u::vector<llvm::SUnit*, std::__u::allocator<llvm::SUnit*>>&, (anonymous namespace)::src_ls_rr_sort&, llvm::ScheduleDAG*) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1871:10
    #5 0x562956e2c4db in (anonymous namespace)::RegReductionPriorityQueue<(anonymous namespace)::src_ls_rr_sort>::pop() third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1905:16
    #6 0x562956e1cc94 in (anonymous namespace)::ScheduleDAGRRList::PickNodeToScheduleBottomUp() third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1472:70
    #7 0x562956e1aaae in ListScheduleBottomUp third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:1625:17
    #8 0x562956e1aaae in (anonymous namespace)::ScheduleDAGRRList::Schedule() third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:383:3
    #9 0x562956e30037 in llvm::ScheduleDAGSDNodes::Run(llvm::SelectionDAG*, llvm::MachineBasicBlock*) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGSDNodes.cpp:64:3
    #10 0x562956dfdb44 in llvm::SelectionDAGISel::CodeGenAndEmitDAG() third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1135:16
    #11 0x562956dfb250 in llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator_w_bits<llvm::ilist_detail::node_options<llvm::Instruction, false, false, void, true, llvm::BasicBlock>, false, true>, llvm::ilist_iterator_w_bits<llvm::ilist_detail::node_options<llvm::Instruction, false, false, void, true, llvm::BasicBlock>, false, true>, bool&) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:874:3
    #12 0x562956dfa88f in llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1897:7
    #13 0x562956df5dcb in llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:614:3
    #14 0x5629560fa5a5 in (anonymous namespace)::X86DAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) third_party/llvm/llvm-project/llvm/lib/Target/X86/X86ISelDAGToDAG.cpp:189:32
    #15 0x562956df2d44 in llvm::SelectionDAGISelLegacy::runOnMachineFunction(llvm::MachineFunction&) third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:374:20
    #16 0x5629568daf48 in llvm::MachineFunctionPass::runOnFunction(llvm::Function&) third_party/llvm/llvm-project/llvm/lib/CodeGen/MachineFunctionPass.cpp:94:13
    #17 0x562958bf0ea1 in llvm::FPPassManager::runOnFunction(llvm::Function&) third_party/llvm/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1406:27
    #18 0x562958bfb121 in llvm::FPPassManager::runOnModule(llvm::Module&) third_party/llvm/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1452:16
    #19 0x562958bf1a06 in runOnModule third_party/llvm/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1521:27
    #20 0x562958bf1a06 in llvm::legacy::PassManagerImpl::run(llvm::Module&) third_party/llvm/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:539:44
    #21 0x562958bfb88a in llvm::legacy::PassManager::run(llvm::Module&) third_party/llvm/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1648:14
    #22 0x562955ad84c9 in llvm::orc::SimpleCompiler::operator()(llvm::Module&) third_party/llvm/llvm-project/llvm/lib/ExecutionEngine/Orc/CompileUtils.cpp:50:8
    #23 0x562955b21103 in decltype(auto) llvm::orc::ThreadSafeModule::withModuleDo<llvm::orc::IRCompileLayer::IRCompiler&>(llvm::orc::IRCompileLayer::IRCompiler&) third_party/llvm/llvm-project/llvm/include/llvm/ExecutionEngine/Orc/ThreadSafeModule.h:136:12
    #24 0x562955b20c9f in llvm::orc::IRCompileLayer::emit(std::__u::unique_ptr<llvm::orc::MaterializationResponsibility, std::__u::default_delete<llvm::orc::MaterializationResponsibility>>, llvm::orc::ThreadSafeModule) third_party/llvm/llvm-project/llvm/lib/ExecutionEngine/Orc/IRCompileLayer.cpp:32:22
[... cut ...]
    #32 0x562955b325a4 in llvm::orc::InPlaceTaskDispatcher::dispatch(std::__u::unique_ptr<llvm::orc::Task, std::__u::default_delete<llvm::orc::Task>>) third_party/llvm/llvm-project/llvm/lib/ExecutionEngine/Orc/TaskDispatch.cpp:23:68
    #33 0x562955adbc9b in llvm::orc::ExecutionSession::dispatchTask(std::__u::unique_ptr<llvm::orc::Task, std::__u::default_delete<llvm::orc::Task>>) third_party/llvm/llvm-project/llvm/include/llvm/ExecutionEngine/Orc/Core.h:1550:26
    #34 0x562955aec933 in llvm::orc::ExecutionSession::dispatchOutstandingMUs() third_party/llvm/llvm-project/llvm/lib/ExecutionEngine/Orc/Core.cpp:2174:5
```

allight, 2024-12-19:
Yeah it seems to be an llvm bug in -O3 x64 codegen. I'll file a bug.

```
[1] % blaze run //third_party/xls/jit:aot_compiler_main -- --output_llvm_ir /tmp/sample.ir.ll --input $HOME/Downloads/sample.ir.txt --llvm_opt_level=0

[0] % blaze run //third_party/llvm/llvm-project/llvm:opt -- /tmp/sample.ir.ll -o /tmp/sample.ir.opt.ll -O3

[0] % blaze run //third_party/llvm/llvm-project/clang -- /tmp/sample.ir.opt.ll -o /tmp/a.out -shared -O3
t801: ch = <<Deleted Node!>>
This target-independent node should have been selected!
UNREACHABLE executed at third_party/llvm/llvm-project/llvm/lib/CodeGen/SelectionDAG/InstrEmitter.cpp:1247!
PLEASE submit a bug report to http://go/clang-crash-bug and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.	Program arguments: /.../blaze-out/k8-fastbuild/bin/third_party/llvm/llvm-project/clang/clang -cc1 -triple x86_64-grtev4-linux-gnu -emit-obj -dumpdir /tmp/a.out- -disable-free -clear-ast-before-backend -main-file-name sample.ir.opt.ll -mrelocation-model static -mframe-pointer=none -fmath-errno -ffp-contract=on -fno-rounding-math -mconstructor-aliases -funwind-tables=2 -target-cpu x86-64 -tune-cpu generic -debugger-tuning=gdb -fdebug-compilation-dir=/.../blaze-out/k8-fastbuild/bin/third_party/llvm/llvm-project/clang/clang.runfiles/google3 -fcoverage-compilation-dir=/.../blaze-out/k8-fastbuild/bin/third_party/llvm/llvm-project/clang/clang.runfiles/google3 -resource-dir /.../blaze-out/k8-fastbuild/bin/third_party/llvm/llvm-project/lib/clang/google3-trunk -O3 -ferror-limit 19 -fgnuc-version=4.2.1 -fskip-odr-check-in-gmf -vectorize-loops -vectorize-slp -faddrsig -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o /tmp/sample-c79e46.o -x ir /tmp/sample.ir.opt.ll
1.	Code generation
2.	Running pass 'Function Pass Manager' on module '/tmp/sample.ir.opt.ll'.
3.	Running pass 'X86 DAG->DAG Instruction Selection' on function '@____sample__main_partition_0'
clang: error: unable to execute command: Aborted
clang: error: clang frontend command failed due to signal (use -v to see invocation)
google3 clang version 9999.0.0 (59890c13343af9e308281b3c76bac425087f4f8a)
Target: x86_64-grtev4-linux-gnu
Thread model: posix
InstalledDir: /.../blaze-out/k8-fastbuild/bin/third_party/llvm/llvm-project/clang
Build config: +assertions, +ubsan
clang: note: diagnostic msg: Error generating preprocessed source(s) - no preprocessable inputs.
```

[sample.ir.ll](https://github.com/user-attachments/files/18202590/sample.ir.ll.txt)