google/xls #1796: Compilation database/compile_commands.json generation fails on linux.

schilkp, 2024-12-16:
**Describe the bug**

Attempting to generate a `compile_commands.json` compilation database using the `./xls/dev_tools/make-compilation-db.sh` script fails with the following error:

```
> ./xls/dev_tools/make-compilation-db.sh
Target //xls/public:libxls.dylib failed to build
Use --verbose_failures to see the command lines of failed build steps.
ERROR: Analysis of target '//xls/public:libxls.dylib' failed; build aborted: Target //xls/public:libxls.dylib is incompatible and cannot be built, but was explicitly requested.
Dependency chain:
    //xls/public:libxls.dylib (501e77)   <-- target platform (@@internal_platforms_do_not_use//host:host) didn't satisfy constraint @@platforms//:incompatible
INFO: Elapsed time: 0.755s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
Traceback (most recent call last):
  File "/home/schilkp/.cache/bazel/_bazel_schilkp/e4cc209945c2029dba88a1739abf7043/external/com_grail_bazel_compdb/generate.py", line 107, in <module>
    subprocess.check_call(build_cmd, stdout=subprocess.DEVNULL)
  File "/usr/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['bazel', 'build', '--override_repository=bazel_compdb=/home/schilkp/.cache/bazel/_bazel_schilkp/e4cc209945c2029dba88a1739abf7043/external/com_grail_bazel_compdb', '--aspects=@bazel_compdb//:aspects.bzl%compilation_database_aspect', '--noshow_progress', '--noshow_loading_progress', '--output_groups=compdb_files,header_files', '--target_pattern_file=/tmp/tmpv4osak4u']' returned non-zero exit status 1.
```

It seems that target platform constraints are causing problems with the dylib (OSX) target for the public API (only?) when bazel is invoked by said script.

The issue bisects to this commit: df5c1f919382af89a7bfe286e82ad2c932f6b25e

As a workaround, removing the `dylib` target from `xls/public/BUILD` allows for the successful generation of the database:

```diff
diff --git a/xls/public/BUILD b/xls/public/BUILD
index 224dec3d2..037244adf 100644
--- a/xls/public/BUILD
+++ b/xls/public/BUILD
@@ -15,7 +15,7 @@
 # Public (exported) XLS APIs.

 # pytype binary only
-load("//xls/public:xls_public_macros.oss.bzl", "libxls_dylib_binary", "pytype_test_test_c_api_symbols")
+load("//xls/public:xls_public_macros.oss.bzl", "pytype_test_test_c_api_symbols")

 package(
     default_applicable_licenses = ["//:license"],
@@ -296,6 +296,4 @@ cc_binary(
     ],
 )

-libxls_dylib_binary()
-
 pytype_test_test_c_api_symbols()

```
I would debug this further but my bazel-foo is limited :)

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the `xls` repo on a linux machine.
2. Run `./xls/dev_tools/make-compilation-db.sh`.

**Expected behavior**
`./xls/dev_tools/make-compilation-db.sh` does not crash and correctly generates a `compile_commands.json` file.

**Environment (this can be helpful for troubleshooting):**
 - OS: Arch Linux
 - Bazel: 7.4.1
 - Python: 3.12.7

 Please let me know if you need more info to repro this - but for me it fails consistently :)

hzeller, 2024-12-16:
This is unfortunately the known current state given the tool we use to create the compilation database.

Your workaround is what I personally do also on Linux :)

There are various tools that allow for creating a compilation database, and the one [we're using](https://github.com/grailbio/bazel-compilation-database) is not maintained anymore. There are newer ones (e.g. [hedronvision](https://github.com/hedronvision/bazel-compile-commands-extractor)) that we have to explore.

cdleary, 2024-12-18:
I think we could also condition the target to be `tags = ["manual"]` based on the platform config, but I haven't worked out how to do that yet.

schilkp, 2025-02-12:
On my machine 5612a3030aa2521240dee4c7f42924842fb083e1 fixed this - Unless this is still broken for someone else this could be closed.

@hzeller on this note: Thank you so much for fixing compilationdb support - Made my workflow significantly easier again!

closed 2025-02-12 by hzeller

schilkp, 2025-02-12:
> Sorry for about a week of no compilation DB at all after switching to MODULE.bazel. Hopefully this is now a more future-proof solution.

No worries. Interestingly this works better for me than the previous solution for tablegen stuff: 

When regenerated, bazel places tablegen outputs at a new path inside `~/.cache/bazel`. The old compiledb solution had these paths hardcoded, so any changes in tablegen'd files was not visible to clangd/clang-tidy until the db was regenerated. Since this now works on paths relative to `bazel-bin`/`bazel-out`, it seems to always pick up the latest version.

I haven't quite worked out which/how one of the crazy `bazel-out/../../../*` paths in `compile_flags.txt` allows clang tooling to find those files, but I have wiped all clangd caches on my machine and it continues to work so I am not complaining ;)

Thanks again!