google/xls #1810: asap7 model for umulp looks wrong

proppy, 2024-12-20:
**Describe the bug**
`umulp` characterization show more delay than `a * b` for asap7.

**Expected behavior**

`umulp` should take less time that `a*b`

**Screenshots**
`a * b` yield the following delay preview:
[image]

`umulp(a, b)` yield the following delay preview:
[image]

proppy, 2024-12-20:
asap7 8-bit umul:
https://github.com/google/xls/blob/7e3d4e81cc669912b4c06094de4a8a6584925c9b/xls/estimators/delay_model/models/asap7.textproto#L18732-L18746
asap7 8-bit umulp:
https://github.com/google/xls/blob/7e3d4e81cc669912b4c06094de4a8a6584925c9b/xls/estimators/delay_model/models/asap7.textproto#L19306-L19320

proppy, 2024-12-20:
Looks like this is the case too for sky130 too:

8-bit umul:
https://github.com/google/xls/blob/7e3d4e81cc669912b4c06094de4a8a6584925c9b/xls/estimators/delay_model/models/sky130.textproto#L18732-L18746

8-bit umulp:
https://github.com/google/xls/blob/7e3d4e81cc669912b4c06094de4a8a6584925c9b/xls/estimators/delay_model/models/sky130.textproto#L19306-L19320

proppy, 2024-12-20:
looks like this actually come from codegen: umulp is simply substracting a fix offset from the `umul` result:
```
  function automatic [15:0] umulp8b_8b_x_8b (input reg [7:0] lhs, input reg [7:0] rhs);
    reg [7:0] result;
    begin
      result = lhs * rhs;
      umulp8b_8b_x_8b = {8'h43, result - 8'h43};
    end
  endfunction
```

proppy, 2024-12-20:
looks like this was introduced w/ https://github.com/google/xls/commit/af042f3d839f02b3f9f68aaae61def93a1ce3903#diff-a2fc2bd44970c5305b7655bcbce2d4e7f01015e489b78fe35bdc46dcfb3b00bd (and that before that the one of the partial product was just 0).

ericastor, 2024-12-20:
I think the change linked above was intended to be specific to simulation.

Possibly the actual issue is that XLS currently outputs its simulation-only stand-in for umulp when targeting a PDK that does not provide partial-multiplication operations.

meheff, 2024-12-24:
Umulp enables splitting a multiply into a partial product (which produces two values) and the final accumulation. This can enable better scheduling because a pipeline register can appear in the middle of the multiply. However, this is only useful if you are using an IP libraries has a fast partial product and use the mulp_format flag in order to instantiate the IP block:

https://github.com/google/xls/blob/b959a4ebbd4182a57bb584590e5089f510469ebb/docs_src/codegen_options.md?plain=1#L274

Otherwise XLS just emits something like `((X * Y - C), C)` for some constant C for the mulp which is not helpful. We probably want to emit an error if mulp is used without mulp_format being specified, assuming we don't do that already.