google/xls #1841: `impl` erroneously requires second template argument to be `u32`

dank-openai, 2025-01-10:
**Describe the bug**

This does not work:
```
pub struct MyType<FOO: u32, BAR: s32> { stuff: uN[FOO] }
impl MyType<FOO, BAR> {}
```

because BAR is s32. It works if I change it to u32. I really do want a signed value, so this is not acceptable.

Error message, referring to the second instance of BAR:
```
sN[32] vs uN[32]: Dimension BAR must be a `u32` (soon to be `usize`, see https://github.com/google/xls/issues/450 for details).
```

**To Reproduce**

Use the code above, try to compile.

**Expected behavior**

That I can make BAR signed without issues.

erinzmoore, 2025-01-15:
Taking a look at this, but given the state of type system v2, if this proves to be very involved, the fix may wait for v2.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): Type Inference v2 (`xls/dslx/type_system_v2/typecheck_module_v2_struct_test.cc`) and `Parser::ParseParametricBindings` (`xls/dslx/frontend/parser.cc:5203-5248`) support signed and non-`u32` parametric bindings (including `s32` with negative values and `bool`) on structs. (Closed by Gemini)

closed 2026-11-14 by allight