google/xls #1848: parametric arguments that are not used in fields are forgotten before they're needed

dank-openai, 2025-01-13:
**Describe the bug**

```
pub struct SomeType<FIRST: u32, SECOND: u32> { data: uN[FIRST] }

pub fn dostuff<FIRST: u32, SECOND: u32, COMPUTED: u32 = {SECOND}>
    (a: SomeType<FIRST, SECOND>) -> SomeType<FIRST, SECOND> {
    a
}

#[test]
fn test_dostuff() {
    let a = SomeType<u32:4, u32:4> { data: u4:0 };
    let result = dostuff(a);
    assert_eq(result.data, u4:0);
}
```
error:
```
0113: pub fn dostuff<FIRST: u32, SECOND: u32, COMPUTED: u32 = {SECOND}>
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^----^ TypeInferenceError: Parametric expression `SECOND` referred to `SECOND` which is not present in the parametric environment;
```

**To Reproduce**
Copy above and build.

**Expected behavior**
I expect this to compile without trouble, and that `COMPUTED` is a parametric argument equal to `SECOND` (which I would then use later in the parametric section or body of the function).

**Additional context**
There appears to be too-early parametric type erasure. Any parametric types not used to define a field of a struct are erased before function parametric type inference occurs.

This problem appears to be caused by the fact that SECOND is not used to define a field in `struct SomeType`. This can be demonstrated two ways:

1. Change: `COMPUTED: u32 = {FIRST}`. This compiles (because FIRST is used by `data`). Now change `data: uN[SECOND]` and it fails to compile (because FIRST is no longer used by `data`.)
2. (don't do # 1 before doing:) Add another field to the struct, and define the field using `SECOND`, e.g. like `unused: uN[SECOND]`. Resolve the compile errors that result from this new field. Now it compiles.

richmckeever, 2025-01-14:
TI2 will fix this. The function parametrics and struct parametrics will be orthogonal in this example, other than the value being propagated to the struct. Parametric struct support in TI2 is not checked in yet, but it may not be worth fixing this in TI1 unless you need it now.

dank-openai, 2025-01-21:
> Parametric struct support in TI2 is not checked in yet, but it may not be worth fixing this in TI1 unless you need it now.

I figured out a workaround that I think will be sufficient. Thanks.

When might TI2 be released?

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): InstantiateParametricStruct in xls/dslx/type_system_v2/inference_table_converter_impl.cc (lines 2801-2960) and xls/dslx/type_system_v2/solve_for_parametrics.cc now bind explicit struct parametrics and default expressions directly into resolved_parametrics even when a parametric is not referenced by any struct member field. (Closed by Gemini)

closed 2026-11-14 by allight