google/xls #1859: Could not evaluate dimension expression ... to a constant value

dank-openai, 2025-01-16:
**Describe the bug**

Try to build this:

```
pub struct foo<IS_SIGNED: u32 , NUM_BITS: u32> {
    data: xN[IS_SIGNED as u1][NUM_BITS], 
}

#[test]
fn test_instantiate_foo() {
    let _ = foo<u32:0, u32:1> { data: u1:0};
}
```

observe error:
```
Could not evaluate dimension expression `IS_SIGNED as u1` to a constant value.
```

**To Reproduce**

Steps to reproduce the behavior:
Use code above, try to compile.

**Expected behavior**
I expect that I can cast a `u32` to a `u1` to define a type. That is, I expect that casting is constexpr-compatible.

cdleary, 2025-01-17:
@dank-openai struct parametrics are different from function parametrics, `as` is fine for constexpr in a general sense it's just a problem in the context of struct parametrics.

dank-openai, 2025-01-21:
> [@dank-openai](https://github.com/dank-openai) struct parametrics are different from function parametrics, `as` is fine for constexpr in a general sense it's just a problem in the context of struct parametrics.

I see. Thanks for the explanation.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): `type_system_v2` (`xls/dslx/type_system_v2/evaluator.cc`) now evaluates `as` cast expressions inside struct parametric defaults and type annotations as valid constexprs, tested in `xls/dslx/type_system_v2/typecheck_module_v2_struct_test.cc`. (Closed by Gemini)

closed 2026-11-14 by allight