google/xls #1862: Docker build fails tests

widlarizer, 2025-01-16:
**Describe the bug**
Docker build fails tests. This means it's unusable since the test step is the build step so since the step doesn't complete I don't even get a semi-functional xls Docker image to run

**To Reproduce**
As described in README: `docker build . -f Dockerfile-ubuntu-22.04` in a fresh checkout, that was 9fea976f in my case

**Expected behavior**
Successful build

**Screenshots**
```
3625.2 Executed 1661 out of 1661 tests: 1652 tests pass and 9 fail locally.
3625.2 There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these are.
------
Dockerfile-ubuntu-22.04:41
--------------------
  39 |     # Test everything (opt), exclude xlscc for now due to increased build time when
  40 |     # we add Clang.
  41 | >>> RUN bazel test -c opt -- //xls/... -//xls/contrib/xlscc/...
  42 |
--------------------
ERROR: failed to solve: process "/bin/sh -c bazel test -c opt -- //xls/... -//xls/contrib/xlscc/..." did not complete successfully: exit code: 3
```

**Environment (this can be helpful for troubleshooting):**
 - OS: NixOS 25.05beta733374.8f3e1f807051 (Warbler) x86_64
 - But like this is Docker so that shouldn't matter at all

cc @QuantamHD

hzeller, 2025-01-16:
Would be good to see which tests failed. It looks like it mostly passed the tests except 9...

```
 Executed 1661 out of 1661 tests: 1652 tests pass and 9 fail locally.
```

I have never used the docker route, but somewhat related if you want to compile natively on NixOS (so, no docker needed), I am keeping a moving target of shell.nix for my personal use around https://github.com/hzeller/nix-xls-build  ... might or might not work for you.  Lilnk the `shell.nix` and `nix/` directory to your checkout of xls, open a `nix-shell` and build from there.

Last time I was able to fully compile xls, but I suggest to `bazel test -c opt ... --keep_going` so that it won't bail out early.

cdleary, 2025-01-17:
Looks like it's fuzz target related -- unclear why this would be any different from CI though:
```
/xls/fuzzer:fuzz_integration_proc_test                                  FAILED in 2.7s
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/fuzz_integration_proc_test/test.log
//xls/fuzzer:fuzz_integration_test                                       FAILED in 3.3s
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/fuzz_integration_test/test.log
//xls/simulation:testbench_io_test                                       FAILED in 0.2s
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/simulation/testbench_io_test/test.log
//xls/fuzzer:run_fuzz_function_codegen_test                              FAILED in 50 out of 50 in 6.0s
  Stats over 50 runs: max = 6.0s, min = 4.1s, avg = 4.7s, dev = 0.4s
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_3_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_1_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_2_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_4_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_5_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_6_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_7_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_8_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_9_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_10_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_13_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_11_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_12_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_14_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_16_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_15_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_17_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_18_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_19_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_21_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_function_codegen_test/shard_20_of_50/test.log
[... cut ...]
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_proc_test/shard_34_of_50/test.log
  /home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/run_fuzz_proc_test/shard_33_of_50/test.log

Executed 1602 out of 1602 tests: 1593 tests pass and 9 fail locally.
The command '/bin/sh -c bazel test -c opt --test_summary=terse --test_verbose_timeout_warnings -- //xls/... -//xls/contrib/... -//xls/dev_tools/...' returned a non-zero code: 3

````

cdleary, 2025-01-17:
```
external/bazel_tools/tools/test/test-setup.sh: line 428: zip: command not found
Could not create "/home/xls-developer/.cache/bazel/_bazel_xls-developer/970c5c2433bb6038ab152477a024c421/sandbox/processwrapper-sandbox/4567/execroot/com_google_xls/bazel-out/k8-opt/testlogs/xls/fuzzer/fuzz_integration_test/test.outputs/outputs.zip":
```
looks like making zipping up undeclared outputs or some bazel rule in rules_fuzzer -- probably the github actions VM image comes with zip apt-get installed but the bare bones one doesn't. I'll send a PR that should fix this.

PR #1865 (2025-01-18): Fix fuzz tests relying on zip binary in docker image build.
Primarily adds "zip" to the installation -- we were relying on this somehow as a system dependency for our xls/fuzzer test targets, I didn't dig into exactly how.

Add common flags and wildcards I use when doing project wide testing.

Fixes #1862
closed 2025-01-18 by copybara-service[bot]