google/xls #1863: Using apt installed xls is undocumented

widlarizer, 2025-01-16:
**Describe the bug**
The documentation instructs the user how to install XLS binaries on an Ubuntu system with apt. However, it's entirely unclear to me how to do anything with it since XLS is closely tied to bazel. When I tried to build `xls/modules/zstd/BUILD` with bazel after installing the binaries, bazel tried to build XLS from source

**To Reproduce**
Steps to reproduce the behavior:
1. `docker run -it -v.:/work ubuntu:22.04`
2. Follow binary installation steps in the docs
3. `bazel-7.4.1 build` in `xls/modules/zstd/` which is probably an incomplete or incorrect command

**Expected behavior**
Either the command builds the outputs of `xls/modules/zstd/BUILD` or a different kind of command is recommended in the docs that does, maybe one with arguments for specifying where the binaries are?

**Actual behavior**

The build starts pulling llvm sources even though there's valid XLS binaries in `PATH`

```
root@ab730c91e672 /w/x/m/zstd [8]# interpreter_main --version
v0.0.0-6732-gaaaa2432d
root@ab730c91e672 /w/x/m/zstd# bazel-7.4.1 build
Computing main repo mapping:
    Fetching repository @@llvm-raw; starting
    Fetching ...ternal/llvm-raw; Extracting 19032bfe87fa0f4a3a7b3e68daafc93331b71e0d.tar.gz
```

**Environment (this can be helpful for troubleshooting):**
 - OS: Ubuntu 22.04 docker on NixOS 25.05beta733374.8f3e1f807051 (Warbler) x86_64

cc @QuantamHD

allight, 2025-01-16:
Unfortunately I don't think that's really supported right now. If you want to use prebuilts you'd need to patch the [xls/build_rules/xls_toolchains.bzl](https://github.com/google/xls/blob/main/xls/build_rules/xls_toolchains.bzl) file and change all the `_DEFAULT_..._TARGET`s to point to the prebuilt binaries. You'll also need to put the binaries in a place that bazel can see them so under the XLS tree would be eaisest.

However, bazel caching means that once its built once subsequent rebuilds will be quite quick so you might just want to stay building from source.

proppy, 2026-10-28:
Rewored this slightly, now that we have releases https://github.com/google/xls/releases we should be able to document the out-of-tree non bazel usage.

proppy, 2026-10-28:
It's actually already documented in the README here: https://github.com/google/xls#install-latest-release

closed 2026-10-28 by proppy