google/xls #1870: [DSLX] Internal error constexpr value not found for local type alias

cdleary, 2025-01-19:
**Describe the bug**

```dslx
fn get_max<N: u32>() -> uN[N] {
    // Note: there is a temporary grammar limitation that prevents us from
    // directly doing `uN[N]::MAX` so we must use a type alias.
    type OutputType = uN[N];
    OutputType::MAX
}

#[test]
fn show_numeric_limits() {
    assert_eq(get_max<u32:4>(), u4:0xf);
    assert_eq(get_max<u32:5>(), u5:0x1f);
}
```
gives:

```
Error: [ RUN UNITTEST  ] show_numeric_limits
E0119 13:28:40.597487 1905801 run_routines.cc:110] Internal error: NOT_FOUND: No constexpr value found for node `N` (NameRef) @ /tmp/tmpio5sr9b5.dslx:4:26-4:27
[        FAILED ] show_numeric_limits: internal error: NOT_FOUND: No constexpr value found for node `N` (NameRef) @ /tmp/tmpio5sr9b5.dslx:4:26-4:27
[===============] 1 test(s) ran; 1 failed; 0 skipped.
```

**To Reproduce**
Run the DSLX interpreter on the above snippet in a file.

**Expected behavior**
The type alias should work with the N binding in local scope.

**Environment (this can be helpful for troubleshooting):**
 - OS: Ubuntu
 - Versions 24.04.1 LTS

cdleary, 2025-01-19:
At a first cut look the trouble seems to come from bytecode emission, not from typechecking.

mgottscho, 2025-03-31:
+1 to this bug, I just hit it on a similar test case.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @dplassgit): Type System v2 (`xls/dslx/type_system_v2/constant_collector.cc:231-246` and `xls/dslx/type_system_v2/type_annotation_utils.cc:393-413`) resolves `::MAX`, `::MIN`, and `::ZERO` on local type aliases parameterized by function parametrics (`type UN = uN[N]; UN::MAX`), as tested in `xls/dslx/tests/attr_via_local_type_alias_in_parametric.x:17-26`. (Closed by Gemini)

closed 2026-11-14 by allight