google/xls #1877: cc_xls_ir_jit_wrapper fails to resolve entrypoints w/ name conflicts

proppy, 2025-01-23:
**Describe the bug**

When attempting to use `cc_xls_ir_jit_wrapper` w/ https://github.com/google/xls/tree/main/xls/examples/matmul_4x4, the toolchain currently fails to resolve the block IR entrypoint w/ the following error:
```
More than one instance with name: __matmul_4x4__matmul_4x4_0_next
```

This seems to be due the block IR containing a `proc` and  `block` w/ the same name:
```
functions: [],  procs: [__matmul_4x4__matmul_4x4_0_next, __matmul_4x4__matmul_4x4__matmul_0__node_0_next, __matmul_4x4__matmul_4x4__matmul_0__node_1_next, __matmul_4x4__matmul_4x4__matmul_0__node_2_next, __matmul_4x4__matmul_4x4__matmul_0__node_3_next, __matmul_4x4__matmul_4x4__matmul_0__node_4_next, __matmul_4x4__matmul_4x4__matmul_0__node_5_next, __matmul_4x4__matmul_4x4__matmul_0__node_6_next, __matmul_4x4__matmul_4x4__matmul_0__node_7_next, __matmul_4x4__matmul_4x4__matmul_0__node_8_next, __matmul_4x4__matmul_4x4__matmul_0__node_9_next, __matmul_4x4__matmul_4x4__matmul_0__node_10_next, __matmul_4x4__matmul_4x4__matmul_0__node_11_next, __matmul_4x4__matmul_4x4__matmul_0__node_12_next, __matmul_4x4__matmul_4x4__matmul_0__node_13_next, __matmul_4x4__matmul_4x4__matmul_0__node_14_next, __matmul_4x4__matmul_4x4__matmul_0__node_15_next, __matmul_4x4__matmul_4x4__matmul_0__4_4_next], blocks: [__matmul_4x4__matmul_4x4_0_next]
```

**Expected behavior**

Since `cc_xls_ir_jit_wrapper` also includes a mandatory `wrapper_type` attr, it should be able to disambiguish and resolve the right entrypoint.

proppy, 2025-01-23:
Looking at https://github.com/google/xls/blob/3a71d294dbba77062795168fb253fb24c137a6fa/xls/jit/aot_compiler_main.cc#L264 it seems that `wrapper_type` is not passed to the AOT compiler and that it only relies on the name for resolving the entrypoint:
https://github.com/google/xls/blob/3a71d294dbba77062795168fb253fb24c137a6fa/xls/jit/aot_compiler_main.cc#L264

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): GetFunctionBaseByNameOfKind in xls/jit/aot_compiler_main.cc (lines 80-81, 177-191, controlled by --top_type=FUNCTION|PROC|BLOCK) and interpret_interface in xls/jit/jit_wrapper_generator_main.py (lines 36-44, 173-216, controlled by --function_type) now disambiguate entrypoints by FunctionBase::Kind when a proc and block share the same name in a package. (Closed by Gemini)

closed 2026-11-14 by allight