google/xls #1885: Formatter crashes when there are comments in array literals

dank-openai, 2025-01-24:
**Describe the bug**

If one has a comment, on its own line, inside an array literal, the formatter crashes. I suspect 

```
E0124 16:38:30.802721 22622349 json-rpc-dispatcher.cc:104] Method error for 'textDocument/formatting' :absl::container_internal::raw_hash_map<>::at
[Error - 4:38:30 PM] Request textDocument/formatting failed.
  Message: absl::container_internal::raw_hash_map<>::at
  Code: -32603 
```

**To Reproduce**

```
pub const TABLE = [
    //break formatter
(u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0),
(u32:0, u32:0),
];
```

Run the formatter.

**Expected behavior**

It should certainly not crash. In the short term, maybe it should ignore (not format) this particular syntactical block, and format the rest of the file.
Not to be prescriptive about how to format... but since you asked: It should format like this, or similar. It should probably align the comment with the other array elements. It should format the non-comments as if the comment didn't exist (but put the comment back where it belongs):
```
pub const TABLE = [
    //break formatter
    (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0), (u32:0, u32:0),
    (u32:0, u32:0),
];
```

dplassgit, 2025-01-29:
In this case it refuses to format that expression because it knows it would lose the comment. It comes out as a crash, but the error message is something like:

```
INTERNAL: Formatting was skipped because a comment at [location] would be deleted by the formatter: //break formatter
This is probably due to a bug (which may not have been reported yet). To complete formatting, try moving the comment to a different line.
```

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `Formatter::FormatArray` (`xls/dslx/fmt/ast_fmt.cc:673-740`) now checks `comments_.HasComments(array_span)` and formats continuous element runs separated by `FormatCommentsBetween` without crashing, tested in `xls/dslx/fmt/ast_fmt_test.cc`. (Closed by Gemini)

closed 2026-11-14 by allight