google/xls #1904: [enhancement][DSLX] Support `..=` inclusive range so pattern matches can use inclusive max value

cdleary, 2025-02-03:
### What's hard to do? (limit 100 words)

Right now you can't do a pattern easily like: `s8:0..s8:127 => ` with an inclusive maximum value for a type because the 127 is excluded and the type is incapable of expression 128. Rust has `..=` to help with this.

### Current best alternative workaround (limit 100 words)

Add another pattern to match arm with | like `s8:0..s8:127 | s8:127 => `

### Your view of the "best case XLS enhancement" (limit 100 words)

We add `..=` support to pattern ranges as well.

msfschaffner, 2025-04-09:
+1 to adding inclusive range matches. It makes things like address decoding in LUTs etc quite a bit more convenient and would be similar to the Verilog `case inside` range matching functionality where ranges are also inclusive.

huangjd, 2025-05-08:
I have a patch for this, expected to land in a few days

proppy, 2025-05-08:
that would be nice to have this for `for` as well (related: https://github.com/google/xls/issues/1671#issuecomment-2484843623)

huangjd, 2025-05-16:
Implemented.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `Parser::ParsePattern` (`xls/dslx/frontend/parser.cc:2410-2423`) and `Formatter::FormatRange` (`xls/dslx/fmt/ast_fmt.cc:1997-2002`) now support inclusive range patterns (`a..=b`, `peek_is_double_dot_equals`) in `match` arms. (Closed by Gemini)

closed 2026-11-14 by allight