google/xls #1911: Apparent parser sensitivity to the positioning of parametric impls

richmckeever, 2025-02-06:
Using this modified code example from a test in a pending type_system_v2 change:

```
struct S<N: u32> {}
struct T<N: u32> {}

fn f<N: u32>() -> u32 { N + 1 }

impl S<N> {
  const N_PLUS_1_VALUE = f<N>();
}

impl T<N> {
  const N_PLUS_2_VALUE = f<{S<N>::N_PLUS_1_VALUE}>();
}
```

When parsing the S impl, the parser appears to deal out a NameRef to N which points to the binding NameDef in T. However, if you move the S impl to immediately after the counterpart struct, it does the right thing.

dplassgit, 2026-11-11:
This gives a different error now: ParseError: comparison operators cannot be chained

allight, 2026-11-14:
Keeping this issue open (reviewed by @dplassgit): We are not closing this issue because the original reproducer now gives a different error (`ParseError: comparison operators cannot be chained`) despite the parser and type system v2 changes in `xls/dslx/type_system_v2/typecheck_module_v2_struct_test.cc:2177`. (Summarized by Gemini)