google/xls #1918: Build Error

siddharth0998, 2025-02-08:
I am new to this and i am trying to build this repo from sources in my mac but i am facing below error. anyone can help me with this?
i am runing bazel build -c opt xls/... this command.

[image]

ritikrajput8660, 2025-02-08:
Troubleshooting Steps:

1 Check for Missing Files:
Ensure that the file MachineApple.cc exists in the src/sta/util/ directory of your project.

2 Case Sensitivity Issue:
If the file exists but is named differently (like machineApple.cc), correct the include directive or rename the file to match the case exactly.

3 Verify Build Configuration:
Open the BUILD.bazel file and ensure that MachineApple.cc is correctly listed in the srcs attribute of the build rule.

4 Clear Bazel Cache:
Sometimes clearing Bazel's cache helps:

bash
Copy
Edit
bazel clean --expunge
Re-run the Build:
Attempt the build again:

bash
Copy
Edit
bazel build //path_to_target
Check Dependencies:
Ensure all required dependencies are fetched and available:

bash
Copy
Edit
bazel fetch ...

cdleary, 2025-02-08:
@siddharth0998 I wouldn't start by building the "full wildcard" of `xls/...`, as you can see it's dragging in some large dependencies that are tripping things up like OpenROAD.

I would start by trying to build particular targets of interest, e.g. particular binaries you wanted to build/try. You may have seen OS X is not a supported platform in google/xls repo CI so nothing is /guaranteed/ to build. In practice, however, a subset of targets can probably be built without too much trouble. If you can say what part of XLS is most interesting to you to I can suggest targets to try.