google/xls #1930: trace_fmt should be support in quickcheck based test

proppy, 2025-02-14:
**Describe the bug**
Currently it doesn't seems that trace_fmt is supported in quickcheck test functions (or function called by quickcheck test functions).

**To Reproduce**
```
fn trace_fmt_bool(b: bool) { trace_fmt!("trace_fmt_bool: {}", true); }

#[quickcheck]
fn prop_test_trace_fmt() -> bool {
    trace_fmt!("trace_fmt_quickcheck: {}", true);
    trace_fmt_bool(true);
    false
}
```

**Observed behavior**

```
[ RUN QUICKCHECK        ] prop_test_trace_fmt cases: test_count=default=1000
...
0486:   fn trace_fmt_bool(b: bool) { trace_fmt!("trace_fmt_bool: {}", true); }
0487:   
0488:   #[quickcheck]
       ______________^
0489: | fn prop_test_trace_fmt() -> bool {
0490: |     trace_fmt!("trace_fmt_quickcheck: {}", true);
0491: |     trace_fmt_bool(true);
0492: |     false
0493: | }
      |_^ FailureError: The program being interpreted failed! Found falsifying example after 1 tests: []
[                FAILED ] prop_test_trace_fmt
[=======================] 8 quickcheck(s) ran.
```

**Expected behavior**
trace_fmt messages are printed

proppy, 2025-02-19:
Note this also fail in `prove_quickcheck_main`
```
[ RUN QUICKCHECK        ] some_quickcheck_prop
E0219 15:22:25.328557 2830867 run_routines.cc:115] Internal error: OK
```

cdleary, 2025-02-23:
PRs that fix these two things are https://github.com/google/xls/pull/1947 and https://github.com/google/xls/pull/1946

cdleary, 2025-02-23:
Oh actually its just the second post that's fixed by the PRs -- I _imagine_ for the OP issue the `trace_fmt` are translated into XLS IR and run, but then we discard the messages instead of emitting them for every sample that the quickcheck presents.