google/xls #1939: Casting array to its own type is not allowed (+ bad error message)

mikex-oss, 2025-02-19:
**Describe the bug**
It seems you can cast scalar values to their own types, but XLS will error out with an unhelpful error message if you try to do the same with arrays.

```
Error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/diagnostics/format_type_mismatch.cc:250) !data.mismatches.empty() FormatTypeMismatch; type mismatch info not constructed correctly for types array vs. array -- we got no mismatches when zip...
=== Source Location Trace: ===
xls/common/status/status_builder.cc:197
xls/dslx/diagnostics/format_type_mismatch.cc:250
xls/dslx/diagnostics/maybe_explain_error.cc:42
xls/dslx/parse_and_typecheck.cc:102
xls/dslx/interpreter_main.cc:198
__main__.XlsRuntimeError('Error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/diagnostics/format_type_mismatch.cc:250) !data.mismatches.empty() FormatTypeMismatch; type mismatch info not constructed correctly for types array vs. array -- we got no mismatches when zipping the types; lhs: uN[32][4] rhs: uN[32][4]\n=== Source Location Trace: ===\nxls/common/status/status_builder.cc:197\nxls/dslx/diagnostics/format_type_mismatch.cc:250\nxls/dslx/diagnostics/maybe_explain_error.cc:42\nxls/dslx/parse_and_typecheck.cc:102\nxls/dslx/interpreter_main.cc:198\n')
```

**To Reproduce**
Steps to reproduce the behavior:

Compile the following DSLX code:

```
fn foo() {
  let x = u32:1;
  let _y = x as u32;  // OK
  
  let indices = u32[4]:[0, 1, 3, 4];
  const DATA_W = u32:32;
  let _data = indices as uN[DATA_W][4];  // Error
}
```

**Expected behavior**
If you try to do the same with a different size element bitwidth, you get a more reasonable error:

```
0011:   let indices = u32[4]:[0, 1, 3, 4];
0012:   const DATA_W = u32:64;
0013:   let _data = indices as uN[DATA_W][4];
~~~~~~~~~~~~~~~~~~~~^----------------------^ XlsTypeError: Cannot cast from expression type uN[32][4] to uN[64][4].
Type mismatch:
   uN[32][4]
vs uN[64][4]
```

Side note: this syntax would be useful to avoid a for loop (#1671), at least until (perhaps even with?) map with lambdas (#1542).

cdleary, 2025-02-19:
Note that the internal error is just flagging that a type error was created with identical types (so it probably shouldn't have been a type mismatch error, as your OP points out)

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): Although casting an array to its own type no longer crashes on `XLS_RET_CHECK` in `xls/dslx/type_system_v2/validate_concrete_type.cc:337-356`, it still gives a bad error message (`TypeInferenceError: Cannot cast from type uN[32][4] to type uN[32][4]`), so we are not closing this issue. (Summarized by Gemini)