google/xls #1940: [DSLX:LS] Ansi color codes don't render in vscode problems pane

cdleary, 2025-02-19:
**Describe the bug**
The dslx_ls produces ANSI escapes in the problems pane in VSCode.

**To Reproduce**
Install DSLX vscode extension
Point at dslx_ls installation
Make a type error where we're displaying member type diffs (e.g. a component within a tuple)
Observe the unredered ANSI escapes

**Expected behavior**
I think somehow other languages provide color coding? But if not we probably need to plumb a "can do ANSI escapes" from the outside world to the error rendering.

**Screenshots**
[image]

cdleary, 2025-02-23:
We do strip ANSI escapes when it's not a tty https://sourcegraph.com/github.com/google/xls@49cff5c1c03aad0e93f0459917b3f3268afb98ec/-/blob/xls/dslx/error_printer.cc?L83-85

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): While `AppendDiagnosticFromStatus` in `xls/dslx/lsp/language_server_adapter.cc:70-87` extracts plain uncolored `err.message` text via `GetPositionalErrorData`, we want actual manual verification of the fix in the VSCode problems pane before closing this issue. (Summarized by Gemini)