google/xls #1941: Add clamped_clog2 parallel to clog2 for parametric bitwidths

mikex-oss, 2025-02-19:
### What's hard to do? (limit 100 words)

Currently, `std::clog2` is special cased to return a parametric width that is constexpr for doing things like:

https://github.com/google/xls/blob/4accf99e348821a23687be5de10e8f1c0bdfcafa/xls/dslx/tests/constexpr_clog2.x#L19-L21

See https://github.com/google/xls/blob/4accf99e348821a23687be5de10e8f1c0bdfcafa/xls/dslx/type_system/deduce.cc#L125-L136

However, commonly in hardware, you want the bit width to cover the case where the thing may have only one entry depending on parameter values, i.e. "clamp" the result to 1 if the input is `<= 1`.

It's easy to write such a function, but such a function can't be used in parametrics. You get the error:

```
TypeInferenceError: Could not evaluate dimension expression <USER_FUNCTION> to a constant value.
```

### Current best alternative workaround (limit 100 words)

Not aware of any workaround.

### Your view of the "best case XLS enhancement" (limit 100 words)

It would be good to at least cover this common scenario and special case some form of `std::clamped_clog2` as well.

More generally, it would be nice to be able to define constexpr functions in DSLX as a user.

cdleary, 2025-02-23:
Not sure what you mean on the last point -- all functions in DSLX are capable of being used in constexpr context.

```
fn my_add_one(x: u32) -> u32 {
    x + u32:1
}

fn p<N: u32, DERIVED: u32 = {my_add_one(N)}>(x: uN[N]) -> uN[DERIVED] {
    x as uN[DERIVED]
}

fn main(x: u8) -> u9 {
    p(x)
}

#[test]
fn test_main() {
    assert_eq(main(u8:42), u9:42);
}

const MY_THING: u32 = u32:42;
const MY_OTHER_THING: u32 = my_add_one(MY_THING);
const_assert!(MY_OTHER_THING == u32:43);
```

cdleary, 2025-02-23:
Oh, I see -- you're saying "it has to be defined in the current module because colon refs are not fully available in the parametric sub-AST". The good news is the whole parametric sub-AST should be deleted with TIv2. cc @richmckeever maybe we want to make a label for "this will be fixed by TIv2".

mikex-oss, 2025-02-24:
@cdleary ah yes, I was trying to add this to a common lib that is imported.

mikex-oss, 2025-02-24:
Update: even after moving it to the current module, I still hit a type inference error within the type of an impl const. Updated the OP to reflect the two usage issues.