google/xls #1956: Simplify udiv/umod when dividend is constant to N-way mux for "small" N

mikex-oss, 2025-02-26:
### What's hard to do? (limit 100 words)

For udiv and umod with a constant dividend, the operations can be replaced by a mux that depends on the divisor. While this can be done manually in DSLX, it can be cumbersome particuarly if you need to handle multiple constant dividends in the same fashion.

If you don't implement this manual transform, then 1) some XLS delay models don't support udiv/umod delay estimates and report an error (https://github.com/google/xls/commit/4261922235bb8ba1aca22e713f5570db96913dd9); 2) if it is supported, the delay estimate used is probably overly pessimistic since hardware synthesis does make this optimization. 

@ericastor pointed out that in order for XLS to know the threshold at which this no longer becomes profitable (because the selector becomes too wide), it would need to incorporate some area estimator.

However, given that dividers are notoriously expensive in hardware, we should be able to be reasonably conservative in choosing when to apply this transformation and still benefit. 

### Current best alternative workaround (limit 100 words)

You can write a manual `match` to express the lookup table for the limited set of possible results instead of just using `/` and `%`. 

### Your view of the "best case XLS enhancement" (limit 100 words)

Support this as an XLS optimization pass.

allight, 2025-02-26:
One complication is that the area/delay model for things like this would either be very inaccurate or extremely complicated since the actual values of any constants matter a lot. Eg no matter the bitwidth `2 / X` is at most a mux of 3 values while higher dividends have way more. The exact range of the divisor matters a lot too for similar reasons.

mikex-oss, 2025-02-28:
@meheff had some thoughts on this during a meeting we had yesterday.

meheff, 2025-02-28:
I think for div and mod we don't really need an area model. These ops should never hit codegen so we always want to specialize if it is at all reasonably possible, and if it is not possible it will be an error in codegen. So setting some reasonable bound on the size of the lut (8 maybe?) and then always converting to a lut if it is less than this is probably a good idea.

ericastor, 2025-02-28:
"should never hit codegen"? I'm surprised to hear that we *never* want to be able to implement integer division, though I'd be on board with warning on it. (After all, I'd assume that for really small div/mod operations, the logic synthesis tools downstream are capable of choosing between a LUT and another implementation... but maybe that's a bad assumption?)

Regardless, I am on board with getting this working faster (or when optimizing without a known target) by setting some simple heuristics.

closed 2025-03-04 by copybara-service[bot]