google/xls #1986: Missing module docstring error on PR import

meheff, 2025-03-06:
When importing PR https://github.com/google/xls/pull/1984 internal check complained about a missing module docstring:

```
CheckLint/0: Warning (//depot/[google3/third_party/xls/METADATA:74:3](http://google3/third_party/xls/METADATA?l=74))
CheckLint found errors.
Missing module docstring [missing-module-docstring]
	//depot/[google3/third_party/xls/dev_tools/check_cpp_includes.py:1:1](http://google3/third_party/xls/dev_tools/check_cpp_includes.py?l=1)
```

@cdleary FYI

cdleary, 2025-03-06:
@meheff I think we need to cultivate the set of pylint waivers to use for OSS developers? As you can see in the PR I already used the waiver for docstrings for functions `# pylint: disable=missing-function-docstring`, but I can't see the list of lint waivers I might want to consider using. What is the module level docstring one?

cdleary, 2025-03-06:
Seems like it's:

`# pylint: disable=missing-module-docstring`

But I don't see that in the repo yet so would be good to confirm 👍

mikex-oss, 2025-03-06:
> I think we need to cultivate the set of pylint waivers to use for OSS developers? 

@cdleary any reason not to just follow the style guide with these docstrings per https://google.github.io/xls/contributing/#code-style?

cdleary, 2025-03-06:
@mikex-oss The problem is nobody outside of Google can run the linter to see what it is flagging, unless I'm missing some way to run pylint as an OSS contributor.

mikex-oss, 2025-03-06:
IIUC, this should work in OSS with https://pylint.org/#install and the pylintrc file linked at the end of https://google.github.io/styleguide/pyguide.html#21-lint.

meheff, 2025-03-06:
I'm guessing pylint could be configured to run as a pre-commit script?

cdleary, 2025-03-07:
That makes sense if it's the same one we run internally -- I was remembering all the g-* rules that I thought were based on internal only verisons, if it's all available in OSS that's ideal. I can poke at this next week.

PR #1987 (2025-03-17): [dev_tools] Flag when a package has no top in `check_ir_equivalence_main`.
Previously this would segfault.

Also adds pylint to the pre-commit yaml; this effectively fixes #1986 as we will now see pylint errors via the pre-commit hook
closed 2025-03-17 by copybara-service[bot]