google/xls #2046: Result miscompare in optimization pass

erinzmoore, 2025-04-21:
Repro: `bazel test -c opt  //xls/fuzzer:run_crasher_test_2025-04-20_ad5c`

From log:

```
Exception when running sample: INVALID_ARGUMENT: SampleError: Result miscompare for sample 1:
args: bits[28]:0xaaa_aaaa; [bits[18]:0x2_aaaa, bits[18]:0x2_2aaa, bits[18]:0x2_8a00, bits[18]:0x3_ffff, bits[18]:0x0, bits[18]:0xaa8a, bits[18]:0x2_ea1a, bits[18]:0x0, bits[18]:0x3_8a22, bits[18]:0x1_a6aa, bits[18]:0xba9a]; bits[45]:0x1155_1752_b972
evaluated opt IR (JIT), evaluated opt IR (interpreter) =
   (bits[28]:0x1, bits[37]:0x11_5517_52b9, bits[1]:0x1)
evaluated unopt IR (JIT), evaluated unopt IR (interpreter), interpreted DSLX =
   (bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
Error: INVALID_ARGUMENT: SampleError: Result miscompare for sample 1:
args: bits[28]:0xaaa_aaaa; [bits[18]:0x2_aaaa, bits[18]:0x2_2aaa, bits[18]:0x2_8a00, bits[18]:0x3_ffff, bits[18]:0x0, bits[18]:0xaa8a, bits[18]:0x2_ea1a, bits[18]:0x0, bits[18]:0x3_8a22, bits[18]:0x1_a6aa, bits[18]:0xba9a]; bits[45]:0x1155_1752_b972
evaluated opt IR (JIT), evaluated opt IR (interpreter) =
   (bits[28]:0x1, bits[37]:0x11_5517_52b9, bits[1]:0x1)
evaluated unopt IR (JIT), evaluated unopt IR (interpreter), interpreted DSLX =
   (bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
```

Output from `ir_eval_main`:

```
bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
// Evaluating entry function at start of pipeline.
(bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
// Evaluating entry function at start of pipeline.
(bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
// Evaluating entry function after pass: dce
(bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1)
// Evaluating entry function after pass: simp(opt_level<=2)(opt_level>=1)
(bits[28]:0x1, bits[37]:0x11_5517_52b9, bits[1]:0x1)
Error: INVALID_ARGUMENT: Miscompare for input[0] "bits[28]:0xaaa_aaaa; [bits[18]:0x2_aaaa, bits[18]:0x2_2aaa, bits[18]:0x2_8a00, bits[18]:0x3_ffff, bits[18]:0x0, bits[18]:0xaa8a, bits[18]:0x2_ea1a, bits[18]:0x0, bits[18]:0x3_8a22, bits[18]:0x1_a6aa, bits[18]:0xba9a]; bits[45]:0x1155_1752_b972"
  simp(opt_level<=2)(opt_level>=1): (bits[28]:0x1, bits[37]:0x11_5517_52b9, bits[1]:0x1)
  before optimizations: (bits[28]:0x0, bits[37]:0x11_5517_52b9, bits[1]:0x1); [after 'Simplification with opt_level <= 2 when opt_level >= 1' pass, dynamic pass #51]; Running pass #52: pre-inlining passes [short: pre-inlining]; Running pass #52: Top level pass pipeline [short: ir]
=== Source Location Trace: ===
third_party/xls/common/status/status_builder.cc:197
third_party/xls/passes/pass_base.h:465
third_party/xls/passes/pass_base.h:570
third_party/xls/passes/pass_base.h:518
third_party/xls/passes/pass_base.h:355
third_party/xls/passes/pass_base.h:214
third_party/xls/tools/eval_ir_main.cc:533
```

ericastor, 2025-04-22:
I was able to minimize this further - and it looks to be a bug in the reassociation pass. @allight, you might want to take a look?

IR:
```
package sample

top fn __sample__main(x1: bits[28] id=1) -> bits[56] {
  x6: bits[1] = or_reduce(x1, id=8)
  zero_ext.23: bits[28] = zero_ext(x6, new_bit_count=28, id=23)
  x13: bits[28] = umul(x1, x1, id=18)
  literal.28: bits[28] = literal(value=0x8000001, id=28)
  x18: bits[28] = umul(zero_ext.23, x13, id=24)
  ret umul.29: bits[56] = umul(literal.28, x18, id=29)
}
```

Reproduction: `eval_ir_main /tmp/test.ir --optimize_ir --input='bits[28]:0x8000'`

You can verify that the difference happens before/after the reassociation pass, by running:
```
$ opt_main /tmp/test.ir --passes=reassociation > /tmp/test.reassociation.ir
$ eval_ir_main /tmp/test.ir
bits[56]:0x0
$ eval_ir_main /tmp/test.reassociation.ir
bits[56]:0x4000_0000
```

Reassociated IR (after DCE):
```
package sample

top fn __sample__main(x1: bits[28] id=1) -> bits[56] {
  x6: bits[1] = or_reduce(x1, id=8)
  literal.28: bits[28] = literal(value=0x8000001, id=28)
  umul.31: bits[29] = umul(x6, x1, id=31)
  umul.32: bits[56] = umul(x1, literal.28, id=32)
  ret umul.33: bits[56] = umul(umul.31, umul.32, id=33)
}
```

ericastor, 2025-04-22:
If it helps, the evaluation trace before (extracted with some transforms on the coverage textproto):

```
x1: bits[28] = param(name=x1, id=1)
  = 0x00800000
x6: bits[1] = or_reduce(x1, id=8)
  = 0x01
zero_ext.23: bits[28] = zero_ext(x6, new_bit_count=28, id=23)
  = 0x01000000
x13: bits[28] = umul(x1, x1, id=18)
  = 0x00000000
x18: bits[28] = umul(zero_ext.23, x13, id=24)
  = 0x00000000
umul.29: bits[56] = umul(literal.28, x18, id=29)
  = 0x00000000000000
```

and after reassociation & DCE:

```
x1: bits[28] = param(name=x1, id=1)
  = 0x00800000
x6: bits[1] = or_reduce(x1, id=8)
  = 0x01
umul.31: bits[29] = umul(x6, x1, id=31)
  = 0x00800000
umul.32: bits[56] = umul(x1, literal.28, id=32)
  = 0x00800000000400
umul.33: bits[56] = umul(umul.31, umul.32, id=33)
  = 0x00000040000000
```

allight, 2025-04-22:
taking a look

closed 2025-04-22 by copybara-service[bot]