google/xls #2062: formatting should preserve newlines introduced by the user

proppy, 2025-04-26:
**Describe the bug**
formatting should preserve newlines introduced by the user instead of trying to minimize new lines.

**To Reproduce**
Consider the following code (a):
```
    assert_eq(
       downcast(HF16::EXP_SIZE, HF16::FRACTION_SIZE>(
            F32 { sign: u1:0, bexp: u8:0b0000_0000, fraction: u23:0b000_0000_0000_0000_0000_0000 },
            u23:0), HF16 { sign: u1:0, bexp: u5:0, fraction: u10:0 });
```
And introducing new lines for readability (b):
```
    assert_eq(
        downcast<HF16::EXP_SIZE, HF16::FRACTION_SIZE>(
            F32 { sign: u1:0, bexp: u8:0b0000_0000, fraction: u23:0b000_0000_0000_0000_0000_0000 },
            u23:0
        ),
        HF16 { sign: u1:0, bexp: u5:0, fraction: u10:0 }
    );
```
Will result in the code being always reformatted to (a).

**Expected behavior**

I would expect the code to stay formatted as (b).

cdleary, 2025-04-29:
Probably would be most helpful for me to brain dump some context. (Maybe the original design doc could be open sourced as I think it has some of this.) The autoformatter is designed via a pretty printer -- it turns the AST into a reflowable document and the document prints itself into a textual output. It doesn't try to reason about how to push tokens around from their original positions like a token-based auto-formatter would (clang-format is notably a token-based formatter). We needed to pick some heuristics for our policies that seemed reasonably self consistent, and minimizing vertical lines seemed like a programmatic one that had clarity vs needing to debate many choices in many contexts.

It's /possible/ to add lots of policy choices to the autoformatter and a configuration file to select things, but honestly, the ROI for that doesn't seem super compelling IMO. At least if it's not "offensively wrong" output from the autoformatter I don't myself tend to think about spending time on it. It's all reminiscent of that quote from go: "gofmt's output is nobody's favorite, but gofmt is everybody's favorite". We made an autoformatter because it was heavily user requested and way better than having nothing, but you can get to diminishing returns of debating where newlines go quite quickly IMO. I know that people do care about these things though! Just I'm not the right person to take assignment to drive at something like this. Hope that makes sense. 🙏

cdleary, 2025-04-29:
Also worth noting: I think this bug would be more actionable if it was framed as a precise policy update based on the example shown, like: "if an invocation is multi-line with a multi-line argument, each argument should start on a new line". That is a description somebody could update the pretty printer to do as a blanket policy update. "Put newlines where I placed them, don't minimize newlines", if that's the crux of the request, is by contrast too wide a request to be actionable with the way our pretty printer implementation works.

meheff, 2025-04-29:
Personally, I think preserving newlines is probably too strict of a rule to follow. It'd be very limiting in what the formatter could do. Also, newlines introduced by the formatter also become unremovable which bakes in potentially sub-optimal or at least arbitrary decisions by the formatter. FWIW I don't believe clangfmt does this.