google/xls #2085: Expression attribute formatting is a bit funky

grebe, 2025-04-29:
When using the new channel attributes, I'm seeing formatting like the following:

```
let (looooooooooong_name0,looooooooooong_name1) = #[....]
                                                                                                   chan<...>();
```

and think this would be better:
```
let (looooooooooong_name0,looooooooooong_name1) =
    #[....]
    chan<...>();
```

dplassgit, 2025-05-07:
That's an outright bug of the chan is all the way to the right

mikex-oss, 2025-05-07:
@dplassgit I believe the far right indentation is just a copy paste error. The formatter aligns the `chan` with the `#` after the `=`.

dplassgit, 2025-05-07:
Ah, so this is WAI. Alternatively, what would be a heuristic to force the # to be on the next line?

mikex-oss, 2025-05-07:
> Ah, so this is WAI. Alternatively, what would be a heuristic to force the # to be on the next line?

When either the `#[....]` or `chan<...>(); overflow the DSLX line length limit.