google/xls #2137: [enhancement] Add `unreachable!` and `unimplemented!` macros

rw1nkler, 2025-05-07:
### What's hard to do? (limit 100 words)

In DSLX, the `fail!` macro is often used both for unreachable code and for code that isn’t implemented yet. Rust has separate macros for these: `unreachable!()` and `unimplemented!()`. These make the code easier to read and understand, and help you quickly find unfinished or broken parts. It would be helpful to have the same kind of macros in DSLX to make intent clearer and make the codebase easier to work with.

### Current best alternative workaround (limit 100 words)

Use `fail!` macro with a correct label and the `false` as a predicate:

```rust
enum MyEnum: u2 {
  A = 0,
  B = 1,
  C = 2,
}

#[quickcheck(exhaustive)]
fn qc(x: MyEnum) -> bool {
    match x {
        MyEnum::A | MyEnum::B | MyEnum::C => true,
        _ => fail!("impossible_value", false),
    }
}
)";

### Your view of the "best case XLS enhancement" (limit 100 words)

Adding `unreachable!()` and `unimplemented!()` macros would make DSLX code easier to read and understand. It would also help people find and track incomplete code or logic errors. `fail!` could then be used only for real failure conditions.

mikex-oss, 2025-05-07:
FYI, for `match`, you no longer need a default arm that is unreachable. 

https://google.github.io/xls/design_docs/dslx_pattern_exhaustiveness/

rw1nkler, 2025-05-08:
Thank you @@mikex-oss for this information! That said, I still believe there is value in adding the `unreachable!()` macro. In some cases a value might technically fall within a valid range, but logically it should never occur.