google/xls #2138: [enhancement] Add `include_bytes!()` and/or `include_bits!()` macro for importing data from external files

rw1nkler, 2025-05-07:
### What's hard to do? (limit 100 words)

DSLX currently has no way to include data directly from external files. This creates extra complexity when working with large portions of data for test cases. In the future, it could also be useful for initializing RAM memory. Rust's `include_bytes!()` macro allows binary data to be imported as a byte array. DSLX could introduce a similar `include_bytes!()` macro, but going further, it could implement a more general `include_bits<WIDTH: u32>!()` macro.

### Current best alternative workaround (limit 100 words)

The current workaround is to define binary data in a separate file (e.g. as an array), and then import that file into the main module. This works, but it requires converting binary data into DSLX syntax manually or with external tools, which adds complexity and can introduce errors.

### Your view of the "best case XLS enhancement" (limit 100 words)

Introducing a  `include_bytes!()` and/or parameterized `include_bits<WIDTH: u32>!()` in DSLX would allow convenient importing of binary files as arrays. This would reduce the complexity of analyzing code and eliminate the need for manual data conversion.

Possible syntax:
```rust
const TEST_DATA:u8[100] = include_bits<u8>!("data.bin");
```

proppy, 2025-05-08:
Not sure if the testing angle as shown in the `TEST_DATA` example is the best case for that enhancement request, as you would need more machinery around includes to manage a diverse corpus of generated test data (ex: if the numbers of generated data is variable); C++ testing is currently better equipped to do this kind of prodetural testing.

For structured data, `include_bytes` would be combersome until we get something like #1698 and `include_bits` would require to assign each of the includes to a  fields of a struct; we currently have https://github.com/google/xls/blob/main/xls/tools/proto_to_dslx_main.cc which is more complete (and complex); it would be nice for the proposal to compare the two approaches and states if they are complementary or alternatives.

For raw data I could totally see how `include_bytes` would be useful to provide initializion data for things like buffers, memories, roms, luts and microcode; it would be interesting to see if we always want to desugar those into DSLX array or if we want to emit some equivalent Verilog construct (like `readmemh`/`readmemb` ?).

cdleary, 2025-05-14:
I think if you can desugar at DSLX time to an array and it doesn't break the downstream tools then you get to avoid the "how do I get access to the runfiles / populate the flist" challenge that might come up if you tried to project it to readmemh in codegen. Just a thought for what might be easier at first. Then, if the scalability of that were a challenge, we could have the ability to codegen it differently (though at that point the value would be opaque to the XLS IR and optimizer, which we don't really have natural support for yet, i.e. extern const values).